ArduPilot/ardupilot #2773: Time delay when switching flight mode using Mavlink.

migvel, 2015-08-28:
I'm experimenting with obstacle detection using a companion computer, when the object is detected(by a sonar) the companion sends a mavlink_msg_set_mode to the Pixhawk.

The Quadcopter is reacting properly but there is delay(around 600ms seconds) in between when the command is sent and when the UAV actually reacts, that takes a bit longer to stop the copter and therefore gets closer to the object than expected.

Right know when the object is detected the UAV switch's to RTL mode from Loiter. Is that behaviour(timing wise) normal or expected?

rmackay9, 2015-08-28:
No, that's not expected.  Are you sure it's not a delay in the serial connection from the companion computer to the Pixhawk?  Perhaps there's a buffer on the companion computer side that's filling up?

I'm going to close this issue because it's not clear yet whether it's actually an ardupilot issue (unlikely), a hardware or a companion computer issue.

It might be best to discuss this on drones-discuss@googlegroups.com.

closed 2015-08-28 by rmackay9

migvel, 2015-09-07:
After checking more I found that the source of delay is not in the Ardupilot software.

Thanks.

rmackay9, 2015-09-08:
@migvel, ah, excellent.  thanks for the follow up.