ArduPilot/ardupilot #2777: UAVCAN is out of date

pavel-kirienko, 2015-08-28:
UAVCAN driver used in APM is quite outdated, and it is not compatible with the release candidate version of the UAVCAN specification. An up-to-date version can be found in the PX4 master, beta and v1.0 release.

Is there any option to expedite the update process?

magicrub, 2015-08-28:
This is something we really need to update, thanks for the kick-in-the-pants reminder! I'm excited to start using/developing on the PX4ESC Pro v1.6.

priseborough, 2015-12-09:
Is there a patch available so APM developers can do testing with newer CAN ESC's? I have some  Ben Dyer ESC's which i wanted to start doing experimentation on with different thrust control modes and the interaction with TECS.

pavel-kirienko, 2016-01-08:
Is update to a newer version of PX4 somewhere on the roadmap? If yes, what is the ETA? If not, is my understanding correct that in order to see up-to-date UAVCAN stack the necessary changes will have to be backported from PX4?

hsteinhaus, 2016-01-22:
As a user of UAVCAN GPS hardware and a big fan of UAVCAN ESCs I would love to help bringing the UAVCAN stack back on track. 

After looking into the differences since the last upstream merge, I can only second that our best chance would be to move to a newer PX4/Firmware release. @tridge - are there any plans on that for the foreseeable future?

WickedShell, 2016-03-15:
UAVCAN was updated 10 days ago. @tridge can this be closed?

closed 2016-03-15 by magicrub