ArduPilot/ardupilot #2778: Plane: restarting mission in-flight with takeoff

magicrub, 2015-08-28:
When executing a takeoff mission cmd in flight, such as when a mission is reset and a takeoff is first, and TKOFF_THR_MINACC > 0, then the motor is suppressed until it feels a launch force (or the altitude is met). if flying below the target altitude then this will cause the aircraft to crash while the motor stays inhibited.

Proposed new takeoff behavior:
If is_flying()==true, then un-suppress throttle.

Jman841, 2015-08-29:
This sounds exactly like what happened here: http://ardupilot.com/forum/viewtopic.php?f=110&t=13441

I thought this was already implemented, but, if not, it would most certainly explain the behavior that happened to us.

magicrub, 2015-08-29:
Yup, sounds like it to me.

magicrub, 2015-08-29:
See this PR for my solution https://github.com/diydrones/ardupilot/pull/2781

closed 2015-09-08 by magicrub