ArduPilot/ardupilot #2782: Plane: Changing jsbsim behaviour to account for grip/friction on the ground

dave-dave, 2015-08-30:
I've been looking into the odd behaviour of SITL around takeoff and landings.
JSBsim puts a reasonable amount of effort into making it work well, it looks like the reason SITL on the ground is weird is that we only use jsbsim for flying.
In jsbsim, FGGroundReactions pays close attention to how the plane interacts with the ground, but I can't see how ArduPlane references that section of code, or how it should be approached.

Does anyone have suggestions for where I should look next? My google-fu is letting me down.
So far I've been poking around ardupilot/Tools/autotest/jsb_sim/runsim.py and ardupilot/Tools/autotest/pysim/aircraft.py

Naterater, 2019-01-07:
I think this might be less of an issue and more of a discussion topic for discuss.ardupilot.org. Have you made any progress on this @dave-dave?

dave-dave, 2019-01-08:
No, no progress.

IamPete1, 2021-08-14:
Sorry no one was able to help.

closed 2021-08-14 by IamPete1