ArduPilot/ardupilot #2787: Plane: ch7/ch8 option to disable airspeed sensor

geofrancis, 2015-08-31:
i recently lost a bixler 2 after the airspeed sensor failed mid flight, the sensor started reporting 250mph causing the plane to cut the throttle and pull up, stalling the plane. it would be very useful to be able to switch the airspeed sensor off and on using ch7 or 8 the same way you can with a rangefinder.

severinleuenberger, 2015-08-31:
How many seconds to you have in flight to analyze, what is the cause of an unwanted behaviour. 
I suggest, by switching to manual, you could get control as well, because the manual mode does just pass trough the rc-signals. 

But the question is; could APM plane provide a logic to detect, if the airspeed gives still valid values? And if there is a missmatch, give back an imediate warn message, that you should switch to manual. Like "airspeed sensor failure detected, switch to manual". Maybe it is even ok to use "stabilized", I assume, there is no airspeed used for stabilized mode.

geofrancis, 2015-08-31:
something as simple as if the airspeed doesn't change for 5 seconds throw up a BAD AIRSPEED HEALTH error and fly without it.

magicrub, 2015-08-31:
The solution here is not to switch to manual, that will crash most planes
of it just randomly switched during audio without to realizing it. Like geo
said, we need to switch it off.

severinleuenberger, 2015-08-31:
You want to switch it off manually, or by logic from APM? I am not that much a fan of "switching manually". I already lost 3 fixed wing uav's (crashed) by not finding the switch fast enough, or when I found it, my plane was already out of sight in the sky...
So, a switch may be implemented without great effort, but the question is: will it really reduce crashes?
If you fly 100m+ above ground, you maybe have enough time to react. From 50m or less, everything goes quite quickly on a fixed wing. A parachute may be the better solution then...

magicrub, 2015-08-31:
This needs to be done automatically, not manually. See https://github.com/diydrones/ardupilot/issues/1628

closed 2015-08-31 by magicrub