ArduPilot/ardupilot #2790: Copter: Guided mode angle controller using SET_ATTITUDE_TARGET

rmackay9, 2015-09-01:
We should extend Copter's guided mode to allow the GCS/companion computer to control the vehicle using angle targets.  Using the SET_ATTITUDE_TARGET message would probably be best.
http://mavlink.org/messages/common#SET_ATTITUDE_TARGET

a7scenario, 2015-09-24:
+1 for Guided-NoGPS flight mode

rmackay9, 2015-10-12:
This is in master except the Guided-NoGPS isn't there yet.

amybear, 2016-04-06:
I'm interested in this functionality. I'm working on an indoor drone navigation application, and it's frustrating that all of the movement commands seem to rely on the vehicle being in GUIDED or AUTO mode (both requiring GPS), even if GPS is not required for that particular movement command.

magicrub, 2016-04-07:
Are you using optical flow or a vision based tracking system for stability?

amybear, 2016-04-07:
Vision-based, specifically a Google Tango.

rmackay9, 2016-04-08:
@amybear,
if you're up for modifying and compiling the ardupilot code you could try modifying the mode-requires-gps so that Guided doesn't require the GPS.  a more complete solution is for us to add a new Guided-noGPS flight mode that is a cut-down version of Guided mode and only accepts the attitude request.  So much to do, so little time.
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/flight_mode.cpp#L276

squilter, 2016-04-10:
@rmackay9 would guided-nogps support `LOCAL_POSITION_NED` messages also? Unlike attitude messages, it will require optical flow.  But I don't think those need gps.

rmackay9, 2016-04-10:
@squilter, if you've got an optical flow, then the regular Guided mode can be used.  So, no, I don't think LOCATION_POSITION_NED would be supported.

jainakshay91, 2016-04-25:
Yes let's have the GUIDED-NoGPS mode. It makes complete sense with so many people working on indoor drone navigation.
At this point in time, to have such a functionality, is the best bet going to be to modify the ArduCopter base code?

allanm84, 2016-04-26:
@rmackay9 is there a specific reason to use SET_ATTITUDE_TARGET to set an angle instead of setting acceleration in one of the 
SET_POSITION_TARGET commands?  Could APM support an acceleration command?

rmackay9, 2016-04-26:
@allanm84, we don't currently accept the acceleration command although a person or two has asked for it.  It could be added, it's just a day or two of work for someone (including testing).
Both acceleration and angle requests should be accepted because they're, of course, not exactly the same.  In particular the acceleration request would be handled as "stabilized" acceleration request meaning that the controllers would ensure that the desired acceleration is achieved... that require require the GPS be functional while the simpler angle requests wouldn't require the GPS to work.

allanm84, 2016-04-26:
@rmackay9 thanks for the explanation.  RE GUIDED-NOGPS, could instead of adding a new mode we just allow GUIDED to operate with no GPS, but restrict which commands would be accepted?  For instance, when there is no GPS: reject SET_POSITION_TARGET commands but accept SET_ATTITUDE_TARGET.

rmackay9, 2016-08-08:
Guided_NoGPS is in master and is also being released as I write this as part of Copter-3.4-rc2.

closed 2016-08-08 by rmackay9