ArduPilot/ardupilot #2795: Navio RC_Input add S-Bus Support

jmachuca77, 2015-09-02:
S-Bus input on the NAVIO board is not working. What is missing to add this support? I might be able to help out with this.

lucasdemarchi, 2015-09-11:
@jmachuca77 navio uses the PCM hardware to get the time of the rise and fall edges of a single pin. I'm not sure how S-Bus works and if it can use only this information. Do you have the protocol explained somewhere?

Anyway, if you want to take a look, you will probably need to adapt these lines: 
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_HAL_Linux/RCInput_Navio.cpp#L492-L517

lucasdemarchi, 2015-09-11:
@staroselskii any idea?

staroselskii, 2017-04-19:
This is not relevant anymore as most of the users are now using Navio 2 either way.

closed 2017-04-19 by staroselskii