ArduPilot/ardupilot #2811: Plane - Auto Takeoff

SlipstreamFPV, 2015-09-07:
I've got a larger plane that is much easier to launch with 2 hands. It would be good if there was a dedicated "takeoff" flight mode that didn't need mission planning to use. One like the MFD autopilot has. My plane is well tuned and can run fully automatic missions takeoff to landing when planned but a launch mode that just had to be set like any other flight mode would be great. One that continues in the same direction you throw the plane. At the moment I have a switch to ramp up to takeoff throttle add elevator and put the plane in stabilise which works but not as well as auto takeoff in auto mode.

magicrub, 2015-09-07:
That would be a good thing for single-person hand launches.

rmackay9, 2015-09-07:
I don't know much about Planes, but one argument I've heard is that we don't know where the plane should go after it takes off.  Maybe there should be some semi-defined default position that it flies to after takeoff.  It could be 100m above "home" if that's been defined by the user through the GCS or it could be 100m directly above the take-off location.. or 100m forward and up from the take-off locatoin.. I'm not sure where it should be or what options should be made available but perhaps this is the thing that is missing and is blocking this request.

SlipstreamFPV, 2015-09-07:
Thanks for your responses. I think the MFD autopilot will just continue in the direction of the throw on a takeoff slope angle until the pilot interrupts. Maybe a straight takeoff to the users set RTL altitude followed by a normal RTL function would work well.

magicrub, 2015-09-07:
@rmackay9 In Auto mode where a TAKEOFF is the first mission command, after it detects a launch it holds it's heading and climbs with a user-defined fixed throttle, generally a high one, until it reaches a mission-defined altitude where it triggers the next waypoint. There's also constraints on roll angle to reduce low-speed stalls.

In a TAKEOFF-MODE I would imagine it takeoffs with high throttle, fixed heading, and positive pitch to a certain user-defined altitude then loiter forever.

However, if switched to this mode during flight it would suffer from what this PR is trying to fix:
https://github.com/diydrones/ardupilot/pull/2781
For bungee launches we inhibit the throttle until a high forward accel is detected and then it engages the motor. The same would have to be done in this mode. So if you have the bungee-type launch configured and you change to this mode in flight it will cause your throttle to go to zero and you'll drop out of the sky.

SlipstreamFPV, 2015-09-07:
That is a serious issue. Can't the takeoff waypoint just be ignored/skipped if the plane is in flight and the mission is reset?

magicrub, 2015-09-07:
Yup, it sure can and my PR fixes that. Problem is @tridge does not yet
trust the "is_flying" flag enough for it to bypass all safety checks. The
"is_flying" flag is a tricky one, how do you know *FOR CERTAIN* that you're
flying?

SlipstreamFPV, 2015-09-07:
airspeed > min speed = flying?

magicrub, 2015-09-07:
airspeed is determined via pitot tube. What happens when you boot up and it
needs calibration but until you do so it reads 10m/s. How about you're
sitting on the ground pointed into a 10 m/s wind?

Take a look at how it is currently determined, I'd love to hear some new
ideas you come up with.
https://github.com/diydrones/ardupilot/blob/master/ArduPlane/is_flying.cpp

-TomP

WickedShell, 2015-09-07:
I think I'm missing something why is this needed? Is it simply about having
to create a flight plan with one waypoint?

I'm failing to see the difference between just leaving a takeoff waypoint
and a whole flight mode?

severinleuenberger, 2015-09-07:
....if you do not have a mission planner or any groundstation software running, then it is a very good thing. A lot of plane pilots fly just with OSD and without groundstation. I would appreciate this mode on most of my fixed wing drones, not only for the big ones. Even for my smallest 76cm wingspan fixed wing, it ist a cool feature to launch it automatically by "LAUNCH" mode, because twisty small planes are critical on launching for the first 2-3 seconds, when there is some wind. 

Second point is - when you have already a mission planned for your flight, lets say one with a loop who starts again on Waypoint 1, you dont want to have the TAKEOFF waypoint in your mission. 

And after takeoff, I usually first switch to FBWA, then to LOITER - just to see, if everything behaves as it should. THEN and not before I start my missions. Thats a security procedure to ensure to see your UAV again after a longer mission ;)

Best wishes, Severin

severinleuenberger, 2015-09-07:
as discussed here
https://github.com/diydrones/ardupilot/pull/2781
the same logic could be applied; 
"done_takeoff" = true after finished takeoff and "done_takeoff" = false before takeoff with reset on arming seems to be a pretty good solution. 
"done_takeoff" = true should be set automatically after reaching 20m above ground.

severinleuenberger, 2015-09-08:
my comment why 20m above ground;
I fly APM 2.6, 3.1 and pixhawk fixedwing drones in sizes from 500g to 9kg. I recognized jumps up to 10m on baro due to turbulences or what ever it was (not calm weather). So it would be a bad thing, if the "done_takeoff" = true would be reached due to barometric pressure sensor disturbances. 
Therefore I reccomend 20m or even 30m. 

Second thing is; for a fixed wing - 20 to 30m above ground after takeoff is not a lot of space between aircraft and ground (unlike copter). I would say 20m is the minimum height who as to be climbed to be in a safe distance above ground. For me 30m would even fit perfect, but for small planes < 1m wingspan, this could be to high.

WickedShell, 2015-09-08:
Is your pixhawk exposed to airflow or prop wash? I've never seen a jump
greater then 3 meters (this is with airframes in the 5 to 3.6kg range). I
agree though that anything below 60 meters feels low at this size.

severinleuenberger, 2015-09-08:
The jump I recognized was in air with a little fixed wing (76cm wingspan) with APM 3.1 on loiter into some gusty winds.  Some up and down could have been there, but I whatched in my OSD jumps from 10m up and 10m down. When I whatched the model circling in the air, this was NEVER more than 1-2m up and down. Therefore my trust in baro on gusty winds are not the best. 

Even tough - I fly big gliders (RC-style) as well, and while thermaling, I trust my baro quite well. 
But as precaution measure, I would go for a value over 10m.

WickedShell, 2015-09-08:
Any airflow over the baro would be bad as I'm sure you're aware. I only
asked as I've helped someone who was having all sorts of problems due to
external airflow over the baro.

magicrub, 2015-09-08:
well I definitely agree that the 5m should be bumped up to *at least* 20m. I'm ok with 30m too. Also, don't forget that a baro exposed to direct sunlight can't be trusted.

severinleuenberger, 2015-09-08:
...it definitely could be, that the shell of my little plane is not airtight enough, so there could be an airflow who create this bumps. But to have a secure measure for not well built fixed wings (maybe as my little one) it would be good to rise the security height to be achieved on a disturbance safe value.

geofrancis, 2015-09-09:
you could just make the takeoff mode harder to do. like only have it on ch7 or 8 and and to enable it you have to flick the switch back and forward 5 times?

SlipstreamFPV, 2015-10-29:
Any progress on this? I'd be so handy to be able to do a 2 handed launch on some of my bigger planes. I don't see what the difference is between this and a proper auto mode auto takeoff. Just hoping to not have to program a mission everytime I want to do an automatic takeoff like currently.

Should be as simple as adding a flight mode that:

-Has the same pre-flight checks as Auto mode
-Executes an auto takeoff maintaining the direction the plane is thrown
-Continues on thrown heading and climbs to RTL alt
-Once RTL alt is achieved then execute RTL
-Remain circling overhead until mode change

magicrub, 2015-10-29:
This mode is not needed. You can simply do an auto mission that has a
takeoff then an RTL, neither need a lat/long so it can be performed from
any launch location without having to change the mission.

WickedShell, 2015-10-29:
I still have trouble seeing the argument? So if you do a hypothetical auto launch mode and takeoff, once you've hit you're altitude, do you move into the mission or always go RTL? If you move into the mission then you need to know what's in the mission anyways and adding a takeoff waypoint there doesn't add any complexity that I can come up with.

What happens if you were going auto and you reengage auto launch? Do you restart the mission (if you had one?)

My concern is there look to be a large number of edge cases here for no added value over a normal AUTO launch... I'd like to move to close this.

rmackay9, 2015-10-29:
Although I'm not a plane user, I'll add my two cents anyway..
I think a takeoff flight mode makes sense.  It should just takeoff and then circle at a given height waiting for the pilot to give it the next set of commands (that might be to take over manually, it might be to fly a mission, it might be to give it a guided mode waypoint position).
Although planes can't sit in place like Copters, I think they can circle around a given target and that's a reasonable thing to do while waiting for more input from the pilot.

magicrub, 2015-10-29:
I volunteer Randy to write it :)

iskess, 2015-10-29:
What added functionality do we hope to achieve over what we already have? Like Tom said, you just need to upload one mission that has nothing but a Takeoff and it will work anywhere and last forever. It will do exactly as Randy described. Unless we don't want the plane to RTL overhead, and instead automatically create a loiter at the top of the Takeoff climb. That would be nice, but I don't really see a big benefit. There are so many other areas that would benefit more from developer attention.

rmackay9, 2015-10-29:
@iskess, because it's easier for the user.  The user doesn't always want to setup up and run a mission but you can be sure they always want to takeoff.  For the same reason, I think a land-flight mode should also be created but.. again.. I'm a copter person.

SlipstreamFPV, 2015-10-29:
Thanks Randy, that's the support I need. Sometimes I have an old mission still saved in my plane and I don't have a laptop at the field to clear it or add a new takeoff only mission. It would certainly make life easier. 

@iskess: by adding takeoff mode no one would lose any functionality only gain it, so why not have this mode?

iskess, 2015-10-29:
Sure, why not. I just didn't see any advantage, but like you said it doesn't take anything away unless the Devs are adverse to too many modes. 
But in your example, with the old mission in your plane, the Takeoff mode will still work anywhere, just switch out of Auto before you reach the top of the takeoff. I realize you might feel this is scary because it might want to fly to the next waypoint 1000 miles away, but I don't see a risk if your failsafe is RTL rather than continue mission.

severinleuenberger, 2016-01-12:
See the closed other issue from me. Its just the way a lot of new and advanced solutions do it. Would be nice, if anyone could give a hand on implementing this.

WillByrum, 2016-02-22:
I think this would be handy to have as a flight mode for things like casual fpv or los flight. I'm building a pixhawk powered small wing now and it would be nice to be able to launch it and have it loiter overhead while I get my other gear together. I like the idea of a flight mode because I'm not planning any telemetry for this aircraft and I don't to bother with a mission. The user could set the loiter altitude etc. The new parrot disco drone has a similar feature:[http://www.theverge.com/2016/1/4/10711178/new-parrot-disco-drone-announced-ces-2016]

magicrub, 2016-02-22:
Why not have an AUTO mission with takeoff? If that's the only mission item
then it will RTL and loiter above home.

WillByrum, 2016-02-22:
I see where that could work but I see the flight mode as less complicated and more convenient for casual flying.

jcooper21, 2016-03-10:
This is a feature parrot build into the disc, the flying wing hobby grade product where it you launch and it loiters overhead until a command is received. If no command is received it will Autoland.

magicrub, 2016-03-10:
I agree that we will benefit from it, no set-up required by pilot to
takeoff. Pull-requests welcome.

pepevalbe, 2016-04-10:
Hi all.

I'm working on this take-off mode. I've spent some time studying the code and I think I can do a Pull Request soon. 
During a mission in a take off command there are two important variables: Take-off altitude and take-off minmum pitch angle. We need to decide where to get these variables in the new flight mode. We have some alternatives:
* They are hard coded to a standar value (eg: 50 meters and 25 degrees). I don't like this option, it is not the style in ArduPilot.
* We create two new parameters for this mode: TKOFF_ALTITUDE and TKOFF_MIN_PITCH. 
* We use existing parameters: For example we could use ALT_HOLD_RTL for the take-off altitude.

What do you think?

rmackay9, 2016-04-10:
I would vote for two new parameters.  I'd recommend shortening "_ALTITUDE" to just "_ALT" although to be more sure, I'd check other parameters in Plane to see if "_ALT" or "_ALTITUDE" is used more.
TKOFF_PITCH_MIN would be another option for the naming of the 2nd param although what I think you have there is fine.
I'm more a copter person than a plane person so opinions from TomP and Tridge should overrule mine.

Boyang--Li, 2016-04-10:
What I usually do is setting to Auto mode by radio with the Auto_Takeoff to be the first command in the mission, then throw out the plane away and change to FBWA mode by radio. I think this is a convenient method.

magicrub, 2016-04-10:
Thanks for taking this on Pepe, lots of requests for it. I think the param method is the way to go, hard coded values never end up working well.

What happens when the altitude is reached? If we loiter above home then I would suggest reusing the RTL loiter alt.
Takeoff pitch name sounds good, I would suggest defaulting to a lower value such as 10deg.

This will be an auto throttle mode so please place it on the mode list appropriately. It also needs to work for hand, runway and bungee launches.

pepevalbe, 2016-04-10:
What happens when the altitude is reached or when we switch to back to take-off mode after succesfully taking-off? I have two different options in mind:
* Behave as in RTL mode: Return to launch and loiter at ALT_HOLD_RTL altitude as Tom suggests.
* Behave as in Loiter mode. I prefer this one. If during flight you accidentally switch to take-off mode the plane will just wait in the same point instead of coming back.

magicrub, 2016-04-10:
I disagree with option 2, I think loitering at home is much better because after a takeoff the aircraft may be far away from you, maybe even too far for the GCS to communicate to it. Having it return after launch gives you time to work on the GCS while the plane flies in a known location and in a place where if you have a problem its at least much easier to take over manual control.

pepevalbe, 2016-04-10:
I don't think the aircraft would be so far away to lose communication unless you set a really high takeoff altitude. If that was true, what is the point on using automatic takeoff in any mission?
Anyway the idea of this new flight mode is to not to need any GCS for auto takeoff

magicrub, 2016-04-10:
100m at 10deg puts you really far away. It should come back after a takeoff. However, if you switch to this mode while already flying then I would say loitering where you're at is OK even at the current altitude.

pepevalbe, 2016-04-10:
Setting takeoff minum pitch to 10 will make the pitch to be 10deg only at the first stage of the takeoff. Once the aircraft gets enough airspeed the pitch is calculated as always so the plane shuouldn't go so far away.

WillByrum, 2016-04-11:
I would lean towards option 1 (behave as in rtl) because my idea of this mode is an auto launch followed by manual control or a manual check of the aircraft before engaging a mission.

marcmerlin, 2017-06-06:
I'd love this too.
For now, I have a mission (which stops me from having another mission) that does:
1: takeoff go to 0,0
2: go to 0,0 alt 80 loiter indefinitely.
Thankfully it doesn't try to actually go to 0,0, and simply climbs straight ahead to altitude 80 and then loiters.
But having this without needing a mission, would indeed be great.

magicrub, 2017-06-06:
@marcmerlin note that TAKEOFF mission item does not have a lat/lng, it's only altitude that matters. The direction is goes is to hold the heading at the time of launch.

marcmerlin, 2017-06-06:
@magicrub correct, I'm just reading the settings from the flight plan, which show lat/long although as you say they are not used.

magicrub, 2017-06-06:
When using Mission Planner, the column labels show up and change depending on which mission item is highlighted.
[image]

marcmerlin, 2017-06-06:
@magicrub 
If you use windows with mission planner (linux laptop, only use mission planner if I must since I have to spin up virtualbox first), yes. 
With apmplanner, not so much. Worse, it draws a line to 0,0 as part of the mission.
Qgroundcontrol does this too by the way, which is kind of annoying.
[image]

OXINARF, 2017-06-06:
@marcmerlin Have you opened issues in APM Planner and QGroundControl for that? @billbonney is probably busy, but @DonLakeFlyer is usually very fast fixing issues.

marcmerlin, 2017-06-09:
@OXINARF done :)

marcmerlin, 2017-06-11:
@magicrub could I ask you to chime in on https://github.com/mavlink/qgroundcontrol/issues/5272 @DonLakeFlyer would like confirmation that it's normal that a loiter command has GPS coordinates of 0,0 and should be ignored for positioning (i.e. don't draw a line to 0.0 like qgroundcontrol and apmplanner both do)

Naterater, 2018-06-11:
Hey @magicrub, any chance of getting this in for testing soon? I would be happy to test!

magicrub, 2018-06-11:
It's on a short to-do list I have. It will get done in the next couple months.

Naterater, 2018-12-01:
This would still be good to have if you ever have time @magicrub

magicrub, 2018-12-09:
I very much want to do this. Step 1 is to get off my ass and finish the plane onion refactor. Then adding modes will be easy.

HITMAnsOFT, 2019-06-07:
As a workaround that does not require a ground station, this is what we do when we want to take off for a manual flight. As all our missions have a TAKEOFF as a first command, whatever mission there is on the plane, we do:
 - Arm the plane,
 - Put into AUTO,
 - Hand launch,
 - Flip the mode switch into RTL when we have enough altitude,
 - Switch into whatever mode we want once the plane comes back.

magicrub, 2020-07-31:
This was added in Plane Release 4.0.0beta4, 19th October 2019

closed 2020-07-31 by magicrub