ArduPilot/ardupilot #2821: Arduplane.elf problems with make in Windows

kah413, 2015-09-09:
I run make sitl -j4 in the ArduPlane directory and get to the very end with an error of two unknown references to healthy (line 77) in AP_RPM.h.

magicrub, 2015-09-09:
Does the usual SITL script work?

kah413, 2015-09-09:
No it still fails at the previous mentioned line of code.

kah413, 2015-09-10:
Also are you referring to the sim_vehicle script? If I run either of these this is the exact error I get.
 
[image]

magicrub, 2015-09-10:
Have you tried cleaning lately? Also, try updating your git submodules

kah413, 2015-09-10:
I ran make clean in the ardupilot directory and a make px4-clean since those were the only files that showed updating. Still same error. Also I was using git submodule init. Downloaded it fresh on a different computer and am getting same error as well.

ericjswenson15, 2015-09-24:
I'm getting the exact same error unfortunately:/

rmackay9, 2015-09-24:
the -j4 option can't be used with windows I'm afraid.

magicrub, 2015-09-24:
You can use -j when using the sim_vehicles.sh script, but I do it in a
virtual machine on my windows computer so maybe that's why it works.

I thought the -j parallel threads problem was related to nuttx, building
SITL slips that so there shouldn't be a problem even on windows.

ericjswenson15, 2015-09-24:
I patched the problem by going into AP_RPM and forcing the function that has line 71 to return -1.

WickedShell, 2017-02-10:
Unable to reproduce on master (as of a couple of months ago)

closed 2017-02-10 by WickedShell