ArduPilot/ardupilot #2828: Build: add solo build target

rmackay9, 2015-09-10:
It would be good to be able to easily build for Solo by using a make command like:
make px4-v2-solo

All it needs to do for the moment is build

magicrub, 2015-09-10:
What is different about the build?

rmackay9, 2015-09-10:
@magicrub, added above. It looks like I hadn't completed my comment before posting.

magicrub, 2015-09-10:
Ahh, gotcha. Also, maybe it could set the imu3_use default value to be 1?

rmackay9, 2015-09-10:
@magicrub, yes, whatever needs to be defaulted differently could be done.  different PID values, etc.

davidbitton, 2015-10-16:
Are the OreoLEDs on the I2C bus and if so, what is the address? Do the OreoLEDs have their own micro-controller each? Thanks!

rmackay9, 2015-10-18:
@davidbitton, a bit off topic but I'll answer anyway.  Also because this is a dev question please feel free to email drones-discuss@googlegroups.com.  People always love fancy LEDs (no joke!) so I'm sure you'll get feedback there.

So, yes, the oreoleds are on the I2C bus.  There are two I2C buses on pixhawk actually.  We call them "internal" and "external" which makes sense on the Pixhawk1 but on Pixhawk2 it's not so clear what that means.  In any case, I think the OreoLEDs start by default on the "internal" bus meaning we would need to make some small changes to the startup scripts (ardupilot/mk/PX4/ROMFS/init.d/rc.APM and change "oreoled start" to "oreoled start b 1") to make the PX4Firmware oreoled driver start on the external bus.

The lowest level code that talks to the oreoled is here in the PX4Firmware repo: https://github.com/diydrones/PX4Firmware/blob/master/src/drivers/oreoled/oreoled.cpp and in there you'll see that the base I2C address is 0x68.  So for Solo the addresses are 0x68, 0x69, 0x6a, 0x6b

The oreoled hardware has an AtTiny88 on it I think that has an I2C bootloader on it which actually allows sending the latest binary to it via I2C.  Not that it matters but in my view, that's actually the most valuable thing in the AtTiny88 firmware.

davidbitton, 2015-10-18:
@rmackay9 it's even more than the start script. The [ardupilot](http://tro.pe/1NiNy6r) code compiles with `OREOLED_ENABLED` set to 0 while the [http://tro.pe/1NiNHa6](ardupilot-solo) code compiles with `OREOLED_ENABLED` set to 1. For my own use, I'm fine w/ changing 0 to a 1 and recompiling; not so sure others are willing and/or able to do so. However, given the `#ifdef OREOLED_ENABLED`, I guess this could just be a supplied to the compiler with an added build target. Thanks.

squilter, 2016-04-17:
`px4-v2-upload-solo` exists now.

squilter, 2016-04-17:
Oops, I see now that we want a command that will change some `#define`s and then compile. I guess that would be called `px4-v2-solo-upload`, not `px4-v2-upload-solo`

peterbarker, 2017-12-19:
We're happily supporting Solo through various means now...

closed 2017-12-19 by peterbarker