ArduPilot/ardupilot #2829: Loss of compass during flight confuses IMU

pwbecker, 2015-09-10:
Twice I have noticed the following: A loose plug on the external compass line results in the copter suddenly losing the ability to stabilise, to the point where it actually flips rapidly upside down.  (Aside from that, the quad is well tuned and flies well, RTL, Althold, Loiter etc).

I didn't connect the dots at the time, so unfortunately don't have log files.

Does the IMU code handle sudden 'zero' inputs from the compass gracefully?  It seems that it may be failing to distinguish between 'no data from compass' and 'zero magnetic field present' (or perhaps its using the last received data, which would also result in crazy calculated orientations).

APM 2, Arducopter 3.2.1, external Neo 8 compass/GPS

rmackay9, 2015-09-10:
@pwbecker,

Thanks for he report.  if you can reproduce and post logs somewhere we can have a look.

I've just done a quick test with an APM2 on AC3.2.1 and a Pixhawk on AC3.3-rc11 and wasn't able to reproduce the issue.  Hard to say what it is but for now I will close it until we have some more evidence.  This is normal, we use the github issues list only for confirmed issues otherwise it turns into a support forum.

closed 2015-09-10 by rmackay9

pwbecker, 2015-09-12:
Thanks for the attention, and I understand re closing.  A final thought: It may have been the sudden RESTORING of the compass data that caused the flip, as opposed to the loss of it, as I implied above.