ArduPilot/ardupilot #2832: Cannot arm in Loiter with EKF_GPS_TYPE = 3 -> Prearm: need 3d fix

dmirk, 2015-09-11:
i am using the px4flow + lidar light combination at my pixhawk with apm 3.3rc10/rc11.
After i have set the EKF_GPS_TYPE to 3 (optical flow) i cannot arm anymore in loiter, giving me the Message "Prearm: need 3d fix". (i have a gps+compass attached too, which is working fine if i change the EKF_GPS_TYPE back to 0)

(I did the setup like mentioned in the wiki (http://copter.ardupilot.com/wiki/common ... -overview/) and all seems to work ok in the prearm-logs.)

fylyll, 2015-10-23:
Hi,

I have the same problem. did you fix it ? i use arducopter v3.3 stable.
Mine copter can't arming in loiter when EKF_GPS_TYPE = 3.
In addition EKF5.meaRng in log is always 0. Do you have this problem too?

Best regards

dmirk, 2015-10-23:
Hi,
no i could not find any fix in the mean time. Someone told me to disable geofence, but this didnt work either. 
My EKF5.meaRng shows correct values, do you have attached the recommended PulsedLight LIDAR-Lite range finder? (http://copter.ardupilot.com/wiki/common-rangefinder-lidarlite/)

Best regards

fylyll, 2015-10-23:
No, i didn't attache Lidar-lite. So you mean EKF5.meaRng comes from Rangefinder not the Px4flow?
I will try lidar-lite soon.

dmirk, 2015-10-28:
Hi, yes in the wiki for the px4flow -  http://copter.ardupilot.com/wiki/common-px4flow-overview/ - is written:
"Although the sensor has plus a built-in Maxbotix LZ-EZ4 sonar to measure height, this has not been reliable enough over a range of surfaces in testing, so its readings are not used and instead a separate Range Finder such as the Lidar Lite should also be attached to the vehicle."

I have used the lidar light at the PWM input of the pixhawk, but since 2 weeks, i get big spikes in my range readings, so that the copter is falling down or shooting up when using AltHold mode with sonar enabled. 

update: Upgrading to ArduCopter 3.3.1 didn't change anything.

dutchrazor, 2015-10-31:
I have the same problem, Arducopter 3.3.1, Pixhawk, PX4Flow with updated firmware, lidar lite.

I've been through the code and there does not seem to be a provision in the arming section for using optical flow instead of GPS. in motors.cpp mode_requires_GPS is checked, and this is set to 1 for Loiter. I can't find any mention in the PX4Flow section of the code where GPS lock is set to true if the flow sensor is operational.

dmirk, 2015-11-01:
Hi, i just found following in the code:
in control_loiter.cpp the function <loiter_init(bool ignore_checks)> is called at init, which checks:
"if (position_ok() || ignore_checks)"

the function <position_ok()> checks for ekf and the optical flow
"return (ekf_position_ok() || optflow_position_ok());"

in system.cpp the function <optflow_position_ok()>
here the values "filt_status.flags.pred_horiz_pos_rel" and "filt_status.flags.horiz_pos_rel" are checked.
This values come from <inertial_nav.get_filter_status()>

My EKF seemst to look good, only some compass interference when not disabling the internal compass of the pixhawk. Perhaps it is an issue with integration into the EKF?

senzai, 2015-11-18:
Hi, I had the same issue.

I'm not 100% sure if this is the right way to solve this problem, but I made the following changes on top of the v3.3.1 and now I can ARM in loiter without GPS 3D fix as long as EKF_GPS_TYPE is set to 3.

https://github.com/senzai/ardupilot/commit/159696a7ae7108c73e4453320a88648e81fb8bfc

Would appreciate it if someone can review and test this.

rmackay9, 2015-11-18:
I'm planning to include a fix for this in the upcoming Copter-3.3.2-rc2 release which will go out later this week.  This is the change I'm planning to include.  https://github.com/diydrones/ardupilot/commit/9b80ab18ae57dc38f1b91ae5946b4b3c317ac7c8

senzai, 2015-11-18:
Hi Randy-san,

Thank you for a very quick response. Enabling OF in any mode makes a lot of sense.

So I should be able to ARM in one of the non-GPS modes (e.g. Alt-Hold) and expect OF to stabilize the copter like Pos-Hold dose with GPS?

I'll apply it on top of 3.3.1 and see how it works. It seems there are many OF related commits that were made after 3.3.1, so maybe I should just wait for RC2...

rmackay9, 2015-11-18:
@senzai, Copter-3.3.2-rc2 will just have a few patches on top of Copter-3.3.1, it won't have all the other changes that have gone into master in it.  If we did that then we would need to go through months of testing again.
The optical flow sensor is really used in place of the GPS so there won't be any changes to manual modes like AltHold but it should be possible to arm and fly in Loiter without a GPS.  Other modes may also work (RTL, PosHold) but they haven't been tested much.

senzai, 2015-11-18:
Randy-san, thank you for sharing more details about RC2.

OK, so I'm assuming, in addition to the commit you shared above, there'll be some changes to motors.cpp to enable arming in Loiter without GPS 3D fix? That's the only thing my changes above do, but I couldn't find anything in master that takes care of this problem.

Thanks again, and please let me know if there's anything I can help with testing.

rmackay9, 2015-11-19:
@senzai,
As far as I know, the commit I linked above is all that's required to allow arming in Loiter (and other modes) without a GPS.  I think it was the position_ok() always returning false if there was no GPS that was the problem.
P.S. looks like you're getting quite familiar with the firmware, great!

senzai, 2015-11-19:
Randy-san,

Thank you for the clarification.

I was not able to ARM in loiter mode with your changes so I looked at the logs and realized my OF was not returning valid values and position_ok() was returning false in motors.cpp. Since AP_NavEKF.cpp doesn't fill prearm_fail_string for OpticalFlow related errors, motors.cpp was spuriously returning the same "Need 3D Fix" error message as before, but for different reasons.

I'll sort out my OF firmware issue and test it again. In the mean time, I'm trying both the [APM recommended version](https://github.com/priseborough/px4flow/tree/klt_flow) and the [upstream PX4 version](https://github.com/PX4/Flow) of the PX4Flow firmware, which seems to be more up to date and I was wondering if either of these is preferred for use with the latest Copter.

Again, thank you very much for your help!

rmackay9, 2015-11-19:
@senzai, aha!  thanks for finding that.  We will need @priseborough to sort that one out.
I think @priseborough is using the APM recommended version but if you want to try the more up-to-date one and report back that might be helpful.

One more thing is you may need to set the GPS_HDOP_GOOD to 9999 to arm.  That's what I needed to do in SITL at least.

priseborough, 2015-11-19:
The EKF nav_filter_status.pred_horiz_pos_rel will be false unless the EKF has the data it needs to do optical flow. We need to change the messaging in motors.cpp so that the message changes if the GPs pre-arm check is disabled. I'll talk to Randy before making changes as i am not overly familiar with the way the various checks in that file interact.

dutchrazor, 2015-11-19:
Hey all,

I flew with the latest master yesterday, and I was definitely able to arm when OF was active.

Thanks!

senzai, 2015-11-20:
Thank you everyone for your support!

I'm happy to report that I was able to ARM and hover in Loiter mode with OpticalFlow enabled and without GPS signal (indoors), using the following:
- PX4FLOW flashed with binary version downloaded from http://ardupilot.com/downloads/?did=118 (px4flow-klt-06Dec2014.px4)
- Pixhawk flashed with Copter v3.3.1 + https://github.com/diydrones/ardupilot/commit/9b80ab18ae57dc38f1b91ae5946b4b3c317ac7c8
- GPS_HDOP_GOOD set to 9999

I also tried the latest version from https://github.com/PX4/Flow , but OPTICAL_FLOW.flow_comp_m_x etc. in both telemetry and dataflash logs didn't look right. Also, when I built https://github.com/priseborough/px4flow/tree/klt_flow from the source, the values reported by the Copter didn't look right either. (values were too small, biased towards one direction only, less frequently updated, etc.) I couldn't fly with OF in either cases.

So I think I'll stick with the binary PX4FLOW firmware and above settings.

Thanks again!

rmackay9, 2015-11-20:
@senzai, ah, excellent news.  Thanks for working with us and reporting back!

sasanfckgw, 2015-11-21:
@senzai  hello men
i am using the Copter v3.3.1 but as i understand you say that change the Copter v3.3.1 code.i,m a beginner.could you help me how can i make change in the code & upload in pixhawk?
thanks a lot

priseborough, 2015-11-21:
@sasanfckgw The information you want is in http://dev.ardupilot.com

sasanfckgw, 2015-11-22:
@priseborough thank you so much
i upload my pixhawk frameware to V3.3.2rc2 & it don't have error when i switch to Loiter mode (it has Prearm: need 3d fix error in ver 3.3.1 so i upload the 3.3.2rc2 frameware),in 3.3.2rc2 stablize mode is ok,but when i try to fly with Loiter mode my quad not accelerate to stay level & don't efect from radio controller commands.can any one help me,what should i do

priseborough, 2015-11-22:
@sasanfckgw Are you trying to fly using an optical and range finder sensor instead of GPS by setting EKF_GPS_TYPE = 3?  If not, then you have a more general support question and should direct it here:

http://ardupilot.com/forum/viewforum.php?f=111&sid=9687143eb9189ab9fb6adc1cc6ecec2c

If you are trying to use optical flow and range finder for navigation and do not have the ability to compile your own code, then I would recommend you wait for AC3.3.2 which will be out soon. If you can compile your own code, then follow the steps outlined by @senzai a few posts above after following the instruction here:

http://copter.ardupilot.com/wiki/common-px4flow-overview/

sasanfckgw, 2015-11-22:
@priseborough  thanks so much for answering soon 
i use EKF_GPS_TYPE=3 and set GPS_HDOP_GOOD set to 9999 also PX4FLOW flashed with binary version downloaded from http://ardupilot.com/downloads/?did=118 (px4flow-klt-06Dec2014.px4) , but i don't use any finder sensor.i know that for use optical flow ability i should connect a rang finder to pixhawk so you think the problem that my quad can't takeoff in loiter mode is for the rang finder?

priseborough, 2015-11-22:
The wiki says you should use a separate range finder. i will update the wording to use the word 'must' instead of should and maybe highlight it in red.

sasanfckgw, 2015-11-23:
@priseborough  yeh,ok. thank you

M9303105, 2016-01-04:
Hi! All
          I set my px4flow following this page step by step(Firmware AC V3.3.2 rc2,  GPS_HDOP_GOOD 9999, EKF_GPS_TYPE = 3), the first problem I met was that I can't arm my drone under loiter mode, so I just arm it under stabilize mode then turn to loiter mode and increase throttle. Plot of log was good in calibration step, Flow, body and gyro were fitting (plz see the following address had log plot picture), but as I flying the drone in real test. Drone was losing control and plot of log became worse.(plz see the video and log file were also in the following address). Dose anybody know what happened to my PX4flow module? and can help me. What should I do? Thanks!!  

https://www.dropbox.com/sh/8tc993ieeym6tpb/AACmLAIVYyDubP2DUbqBWoxNa?dl=0

lthall, 2018-01-12:
This looks like it is sorted and is extremely old for the EKF 3 now.

closed 2018-01-12 by lthall