ArduPilot/ardupilot #2835: Plane: Normal to Land flight stage transition instability

magicrub, 2015-09-13:
plane: flying wing
Arduplane 3.03 and master

From Hao Chen's drones-discuss post that I have confirmed: https://groups.google.com/forum/m/#!topic/drones-discuss/XbWOJkiAxME
I noticed that my wing tends to drop off throttle sharply right at the transition point between normal flight and the start of auto land phase regardless of whether or not its needed. It's not an issue when the flight path is already aligned with the land wp as since it lends with the descend. However, when there is a sharp turn following the transition the plane tends to compensate the throttle drop off by pitching the plane down and up right before the turn. I tried to tune out this behavior by changing all the TECS_LAND params but so far had no succeeded. Can anyone take a look at the log and help with minimizing the drastic pitch instability right at the transition to auto land please?

In the log you will see that I've put the plane into the landing pattern a number of times to observe and try to correct for the pitch instability. The typical pitch adjustments are much less noticeable during a normal banking/turning maneuver.

[image]

Logs from gitter user @gelazanskas
https://www.dropbox.com/sh/q3z0ijjbkxnp4o2/AADMUnVxM94tb3IhMkHCMTV0a?dl=0

Logs from Hao Chen @haogodent2 :
http://haogodent2.us.quickconnect.to/fbsharing/cOmgFAqE

tridge, 2015-09-15:
The cause of the issue for gelazanskas is that the rangefinder settings are incorrect. It is an analog Lidar, and it seems to give a maximum reading of 39.5m. When it doesn't see a target it gives its maximum range (ie. 39.5m). The RNGFND_MAX_CM is set to 40m, whcih means ArduPilot thinks that 39.5 meter reading is within the Lidars range, and thus valid.
So on landing approach (which is the first time the Lidar is used) the aircraft wants to be at 50m, but the Lidar is saying it is at 39.5m. It thus wants to climb. As the glide slope is recalculated on each loop it just wants to climb more.
The fix is to change both RNGFND_MAX_CM and RNGFND_MIN_CM to the right values. Setting RNGFND_MAX_CM=3800 and RNGFND_MIN_CM=200 would fix the issue.
Meanwhile I am trying to find a way to change TECS so it does something more sensible when it is getting bad Lidar input.

tridge, 2015-09-16:
I have pushed a change to master to try to catch bad lidar readings and disable the lidar

magicrub, 2015-10-21:
This is no longer a problem with https://github.com/diydrones/ardupilot/pull/3013 applied

closed 2015-10-21 by magicrub