ArduPilot/ardupilot #2842: Bug when rover in DGPS. No Logs are saved.

QuintinoDuarte, 2015-09-14:
There is a bug when the rover is in DGPS mode, KMZ log file is not correct when using Mission Planner to convert the log file.
The best way to see this is to create a mission and set the rover in to and out from DGPS mode. The  kmz file only shows the path when the rover is not on DGPS mode.

RoverHunter, 2015-09-14:
I can now confirm this is a Mission Planner bug when using the "Create KML + gpx" tool. I can check the entire mission using the "Review a Log" under GPS/Status 1(NO_FIX), then 3 (GPS_OK_FIX_3D) and finally 4 (GPS_OK_FIX_3D_DGPS), however when in this last mode, no path on the kmz file.

closed 2015-09-14 by meee1

magicrub, 2015-09-14:
What versions of rover and mission planner?

magicrub, 2015-09-14:
Ooh, I see MichaelO just closed the issue, he must be on the case...

RoverHunter, 2015-09-14:
Apparently this problem is solved in the last Beta version of Mission Planner.