ArduPilot/ardupilot #2854: APM Copter Wiki Link Update for MaxSonar

MaxBotixInc, 2015-09-15:
Hello,

We recently had found an error in linking on the APM Copter Wiki. 

I located the error on the following page:
http://copter.ardupilot.com/wiki/common-rangefinder-maxbotix-analog/

Please see the attached image showing the issue with the embedded link.

I appreciate your help in updating this link. 

Best Regards,

Scott Wielenberg
sdwielenberg@maxbotix.com
[image]

MaxBotixInc, 2015-09-24:
MaxBotix Inc., is offering a free sensor to the first user who can complete this requested update. Please contact info@maxbotix.com for complete details and request to speak with Scott. 

The MaxBotix Inc team looks forward to working with you. Have a great day!

rmackay9, 2015-09-26:
Ok, I think this is fixed now.  I've just updated it on the planner version of the page and it should propagate automatically to the copter, plane and rover copies.
http://planner.ardupilot.com/wiki/common-rangefinder-maxbotix-analog/

Nobody expects everyone to know all the intricacies of how we work but just in case you find  more wiki errors, this is where we track the wiki issues: https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues

closed 2015-09-26 by rmackay9