ArduPilot/ardupilot #2857: LSM9DS0 compass driver

raspilot, 2015-09-16:
LSM9DS0 is 9-axis sensor. It looks like someone is working on it because there is a driver in AP_InertialSensor.
But no any LSM9DS0 compass driver could be found in AP_Compass. So I'm curious about that is there a schedule on it?

Since it looks like that LSM9DS0 is an exact combinition of LSM303D and L3GD20, the AP_Compass_LSM303D could be combined to a same driver. And RasPilot (using LSM303D & L3GD20) could benefit from the work too.

proficnc, 2015-09-16:
We use this sensor on the PXF

raspilot, 2015-09-16:
@proficnc Nice! Are you using the compass on it?

proficnc, 2015-09-16:
No, but no reason you cannot... But the HMC 5983 is far better....

raspilot, 2015-09-16:
@proficnc Agree. I'm using  HMC as the external compass, and LSM303D as an onboard secondary compass, just as pixhawk.

lucasdemarchi, 2015-09-25:
@raspilot in a previous iteration of our board we had this sensor. So we fixed it and submitted upstream. We are not using it now, but maybe will use in future.
I think when we were doing it @guludo said it's not only the combination of these 2 sensors in the same package but that there are actually differences among them.

raspilot, 2015-09-25:
@lucasdemarchi Thanks for your response. I checked the datasheet of those three chips. The features, data bus wiring, registers shows LSM9DS0 seems like a combination of LSM303D & L3GD20, except the direction of gyro. 
I tested LSM9DS0 with a breakout board on raspilot. The program also works on LSM303D & L3GD20(Need to rotation the data of gyro). Here is my picture:
[image]

Naterater, 2019-01-07:
This driver exists now, correct?

WickedShell, 2020-12-08:
Closing. I can see a driver in master for this.

WickedShell, 2020-12-08:
Ah, wait I misunderstood, there isn't a compass driver, just the accel/gyro side.

lucasdemarchi, 2020-12-08:
@WickedShell libraries/AP_Compass/AP_Compass_LSM303D.cpp

closed 2021-08-09 by lucasdemarchi