ArduPilot/ardupilot #2859: ardupilot for PX4 Code editing with intellisence!?

thexnightmare, 2015-09-16:
Dear,
I am editing ardupilot code for APM 2.5 on Visual Micro with intellisence (code auto-completetion) without any problems.
How can I do the same for editing the code of ardupilot for PX4?
Thanks

magicrub, 2015-09-16:
I suggest using eclipse.

closed 2015-09-16 by magicrub

thexnightmare, 2015-09-16:
I am using it with the tool-chain, but even though, no intellisence working!?

magicrub, 2015-09-16:
It works, I have it on mine. Click on a folder and press F5 to refresh it.
Then right-click and "Run C/C++ Code Analysis".

thexnightmare, 2015-09-16:
I've clicked on the root (ardulilot) folder, then click F5, then run c/c++ code analysis, and seems the intellisense is working fine now, thanks, however, I am aware of the number of problems that appears.
[image]

magicrub, 2015-09-16:
That's normal. Eclipse can not compile our code it must be done from our
own scripts. There are dependencies in other modules/repos.
Have you gone through the wiki on compiling with Eclipse? There's info
here: http://dev.ardupilot.com/wiki/building-with-eclipse/

I compile using this method here:
http://dev.ardupilot.com/wiki/editing-the-code-with-eclipse/

-TomP

thexnightmare, 2015-09-16:
Yes, I am compiling it normally through these guides, without errors, just was wondering about the high number of errors that appears. Thanks alot