ArduPilot/ardupilot #2879: Log if GPS updates are slow

rmackay9, 2015-09-22:
During Copter-3.3 beta testing one users's GPS started updating slowly (2.5hz instead of 5hz) about 18 seconds before it lost lock completely.

It would have been good if we could log the slow update rate in the dataflash logs as an error to ease support - it took me a while to notice the slow update rate.

We should probably also have reported the GPS as unhealthy to the ground station somehow so that the user was aware there was a problem.

WickedShell, 2016-02-08:
#3580 sends this as a critical status text (for u-blox) but doesn't put it in the DF log. Is this fine or should we also be logging it to the DF?

Because the times when we get the packets are already logged I wonder if it makes more sense to add this as a automatic log analysis check?

OXINARF, 2016-11-03:
@rmackay9 Is this something we still want?
@WickedShell If I'm not mistaken we don't send that message anymore. Should we add a message for all drivers if a minimum update rate isn't kept?

I agree that instead of logging this to DF it should be in the automatic log analysis tools.

WickedShell, 2017-01-13:
We should manage this at the AP_GPS level rather then the driver level, which is what was happening before. I propose that AP_GPS considers if the GPS is maintaining the desired rate and adds a healthy() interface. For simplicity I would recommend that healthy() considers if the primary GPS instance is healthy() rather then one per instance as we should be using this to change the status flags coming out over MAVLink. (A healthy() per instance could be used for logging though).

OXINARF, 2017-01-14:
Sounds good. I agree that, unless the full GPS library is changed to not have a primary instance (which I'm not sure would be a good thing), a *healthy()* interface should report just the primary. But it should have another *healthy(instance)* interface so we can get the healthy status for all GPS instances.

WickedShell, 2017-06-07:
#5749 it the path forward on this one. Healthy is also weird as it sort of semantically implies is_configured()

I'm okay with closing this entirely in favour of #5749

WickedShell, 2017-06-08:
Closing in favor of #5749

closed 2017-06-08 by WickedShell