ArduPilot/ardupilot #2880: Guided mode turn into RTL mode unexpectedly

PenguinCheng, 2015-09-22:
Hello,
I am working on a job that uses an odroid-u3 connected with pixhawk to control my quadrotor. After i ran my code for a while, the pixhawk turned into RTL mode. However, i didn't write the command to turn into RTL mode. In my code, i give velocities every one second and the function is ‘send_ned_velocity()’(vehicle.message_factory.set_position_target_local_ned_encode()). 
I don't why it ran into RTL mode unexpectedly. And in what situation, the pixhawk will run into RTL mode?
could someone help, thak you in advance!

killerlayup, 2015-09-22:
I have the same issue! It really confused me..... T T

magicrub, 2015-09-22:
Have you checked your FS_* params?

rmackay9, 2015-09-23:
This is really a support and/or developer question so I'll close it here.  Sorry to be annoying but otherwise the github issues list becomes a support forum.

If we consider this a developer question, then posting it on drones-discuss is ok.  If it's a Copter-3.3-rc that you're working with, posting this on the Copter-3.3 beta testing thread is also a-ok.

closed 2015-09-23 by rmackay9

PenguinCheng, 2015-09-23:
I am going to check my FS_* params, to see if it's the problem. Thank you for your reply.

PenguinCheng, 2015-09-23:
Thank you for your advice!@rmackay9