ArduPilot/ardupilot #2885: Copter: force the exit form any failsafe with "double flight mode switch"

robustini, 2015-09-23:
A friend has destroyed his quad because unfortunately switched a flight mode half a second after the battery failsafe is activated, continued to fly not realizing that the battery was too low.
The "RTL" in the image are related to "battery failsafe", although for reasons that we are investigating is not properly displayed as "failsafe event".
To avoid this event, which can happen if you're very unlucky (like my friend), i therefore propose to include in the code an option that allows to force the exit from any failsafe only by double flight mode change, who still I think it is even more logical, in this way to force the failsafe exit execute a double switch in the back way where I was flying (es. i'm in Stabilize, to force the exit i switch to Loiter and back to Stabilize).
Another friend mistakenly switched to another flight mode while its S900 was performing an RTL from "battery failsafe" and had the throttle to zero, switching to Stabilize destroyed the drone.

[image]

geofrancis, 2015-10-03:
i just don't use stabilised in anything that absolutely doesn't need it. althold is perfectly good for take-off and landing but without the risk of turning the motors off. i stopped putting stabilised as an option for new pilots because guaranteed one will panic and flip to stabilised with the throttle down and it falls out the sky.

WickedShell, 2019-01-07:
No that doesn't overlap the requested functionality at all, once a failsafe is triggered any mode change action pops you out of it until a more critical one occurs, which has a different handler and means you missed the desired behavior.