ArduPilot/ardupilot #2888: Compass MOT applied even when disarmed/motors not running

andyp1per, 2015-09-23:
MOFS_X etc are applied to the compass values even when the copter is disarmed. On an AUAV-X2/ACPS3 at least this causes bogus compass values to be calculated when running on USB power only because this combination gives spurious amperage metrics when disconnected from the battery (e.g. 16A on my copter). This then causes the "inconsisent compasses" message to kick in.
Ideally the X2 would read 0 on no input, but a software solution would be to only apply the offset when the motors were running or when the copter was armed.

rmackay9, 2015-09-24:
@andyp1per, thanks for the report.
It sounds like it could be considered a hardware issue though.  My IRIS+Pixhawk doesn't display crazy current values when plugged in through USB.

andyp1per, 2015-09-24:
Hi Randy, yes I agree its a hardware issue but seems like not applying offsets when on USB power is a prudent thing to do. I have reported it to the AUAV guys.

WickedShell, 2019-04-09:
Closing this as this is a hardware problem. People do connect the USB while in flight (unfortunately) so we can't afford to simply disable the compensation when USB is present. The problem with USB only power tends to become that the way your power system is setup the current sensor isn't powered, which leads to a floating/non 0 value on the ADC pin.

closed 2019-04-09 by WickedShell