ArduPilot/ardupilot #2895: SR4 parameters don't exist

fnoopdogg, 2015-09-24:
Now that Copter-3.3 supports serial4 for mavlink use, how do we set SR4 parameters?

rmackay9, 2015-09-26:
SR0 ~ SR3 parameters should be used to set the rates or better yet use the MAVLink request_data_stream message if we're discussing a companion computer or GCS that wants to set the rate of the available data streams.
http://mavlink.org/messages/common#REQUEST_DATA_STREAM

closed 2015-09-26 by rmackay9

fnoopdogg, 2015-09-26:
Apologies if I'm getting this confused (doesn't take much), doesn't SR4 refer to Serial 4?  Now that Serial 4 has been added it's very useful for adding more components that we didn't have room for before.  According to the docs the data_stream/request_data_stream is deprecated, and we should use the SRx_ rate parameters instead?