ArduPilot/ardupilot #2896: Flap Control from GCS

Jman841, 2015-09-24:
I thought this was an issue in mission planner, but, was told this may be an Ardupilot issue. Original issue is here: https://github.com/diydrones/MissionPlanner/issues/1013

As of right now, unless I am mistaken, there is no way for me to control the flaps of the plane with the Tx off unless I am flying and do a "Do Change Speed" or "Land" command. The issue is, for some reason with my current setup, if my Tx is powered off, the default action on my plane is flaps down. It would be awesome if there was a way to toggle the flaps from the GCS, or, have a flaps up/flaps down button to control them from the GCS. We sometimes use a joystick to fly, and, I have not figured out how to activate flaps from a joystick either. With the current setup, if I don't use the Tx, there is no possible way for me to retract the flaps.

If I use the "Servo" tab in mission planner and try to toggle it there, it simply makes the servo twitch and return to it's position of full flaps down.

magicrub, 2015-09-24:
> an Ardupilot issue. Original issue is here: diydrones/MissionPlanner#1013

Most rc receivers output a certain default output when the transmitters is
off. This default output is assigned when binding. Try rebinding your
transmitter with the mode and flaps position set to whatever you want it to
output when the transmittet is powered off.

Jman841, 2015-09-24:
That may solve my specific issue, but, I still think it would be nice to have control of the flaps when not using a transmitter for preflight checks, or, if for some reason in flight you need to activate them but don't want to change speed. Maybe for your landing path approach (We do a box shaped pattern to reduce altitude and line up for the final land waypoint) to maintain current speed while descending in preparation for the final "Land" waypoint when they would be automatically deployed?

Naterater, 2019-01-07:
I do see the value in being able to control flaps from a GCS. Is this a limitation of mavlink?

peterbarker, 2019-01-07:
On Sun, 6 Jan 2019, Nathan E wrote:

> I do see the value in being able to control flaps from a GCS. Is this a
> limitation of mavlink?

There's not much reference to them.

Correct thing to do is probably to put it into some parameters in 
MAV_CMD_AIRFRAME_CONFIGURATION - perhaps make parameter 3 a percentage 
flaps to apply?

Note that this may not be trivial; you have to worry about interaction 
with both auto-flaps and flaps-on-a-channel.

magicrub, 2019-01-07:
>you have to worry about interaction with both auto-flaps and flaps-on-a-channel.

perhaps this could be done like RC override via timeout. GCS has to keep sending the flap (or misc override) command every second-ish.

peterbarker, 2024-07-31:
We can do this now with a DO_AUX_FUNCTION mavlink command - and MissionPlanner has a special UI for those.

closed 2024-07-31 by peterbarker