ArduPilot/ardupilot #2899: Stabilize_Init() is not run on start-up.

R-Lefebvre, 2015-09-25:
Since the flight mode initializes as Stabilize, and the mode Init() functions are only run when processing a mode change, if a system boots up in Stabilize mode it will not run the stabilize_init() function, possible missing important functionality, (but there is no real problem now).

rmackay9, 2016-08-29:
bumping to Copter-3.5.

shinjanxp, 2018-02-22:
@rmackay9 @peterbarker I would like to take up this issue. Could you please refer to which files I should look at to solve this? Thank You.

peterbarker, 2018-02-22:
ArduCopters system.cpp in `init_ardupilot()`.  It may be as simple as a call to `flightmode->enter()` - but it needs to be tested!

If you don't have a SITL environment yet - get one :-)

rmackay9, 2018-10-25:
I think @tridge's idea is good and it also consistent with how Rover does it.. and growing consistency between the vehicles is great for us all

int-smart, 2018-10-25:
Hi
I created a Pull request after making the changes and successfully building and testing it for the different plarforms. Here is the pull request number "ArduCopter: Changed the ArduCopter code to include INITIALISING mode …" #9633. Let me know if any changes have to be made.

peterbarker, 2021-07-30:
https://github.com/ArduPilot/ardupilot/pull/15706/files means we now run the relevant init code

closed 2021-07-30 by peterbarker