ArduPilot/ardupilot #2907: IR Lock for Arduplane net landings

Jman841, 2015-09-27:
Would it be possible to add the IR Lock feature of Arducopter to Arduplane? Could we set the sensor forward facing and have it look for an IR Beacon in the center of a net for reliable net landings?

Recently I did some net landings, but, just did it in FBWA and manual control and it seems to be an excellent solution for our mapping missions in certain regions where we don't have access to good place for landing.

iskess, 2015-09-27:
+1!  I'm trying to get reliable net recovery too. A virtual ILS system would be wonderful. Lateral guidance is easier than vertical guidance.

magicrub, 2015-09-28:
+1
I'm game!

rmackay9, 2015-09-29:
I received the same request from a user here in Japan about 2 weeks ago who does just what @Jman841 does.  I'm assuming @Jman841 didn't come to see me 2 weeks ago!

Jman841, 2015-09-29:
Haha while I would love to go back to Japan sometime soon, that was unfortunately not me. 

Reliable net landings can solve a lot of the draw backs of fixed wing without having to make compromises to add a VTOL component. I think this would be an amazing feature to have!

sazzal10, 2016-10-26:
Here to revive this old thread.

I'm looking to do exactly this, use the IR lock feature for net landings. I've have been attempting to modify the Plane code with the IR lock features from ArduCopter. No success yet. (I'm am fairly new to writing code). 

Has any one else experimented with this feature? What is required to use the IR lock function in ArduPlane? I have added what I believe is the required code to the Arduplane source but I'm having some issues compiling. I would love to get some pointers from someone with more experience than me. (I mainly deal with hardware..) At the moment I'm just trying to add the Precision Landing parameter and have the Pixycam recognized by the board. 

Thanks in advance.

magicrub, 2016-10-26:
Here's an additional comment from @sazzal10 in issue https://github.com/ArduPilot/ardupilot/issues/5072 that I closed as duplicate but it has some details we should capture in this thread (such as he wants to do it!) so I thought it's worthwhile to repost the content here:

> Hello guys, I'm new to this great community.
> 
> I'm working on a project that requires precision landing for fixed wing planes.
> Similar to the function already implemented in the ArduCopter, I want to use the Pixycam and a large infrared beacon to land the aircraft into a net. I have used this function on a quad and had no problems making it work great.
> 
> Currently, I have the aircraft flying very well using the Plane 3.6 firmware. My plan is to modify the existing plane firmware with the required ir lock/ precision landing code form the Copter firmware.
> 
> My question is, has anybody experimented with this before? What is required to "merge" the existing "Precision Landing" Copter parameter into the Plane firmware? I understand that the orientation of the sensor will be 90 degrees to whats normal on quads, but at the moment, I only want to add the parameter and compile the firmware with the required modules. (One step at a time)
> 
> I'm new to the ardupilot code but educated on how to compile, edit and upload a custom firmware to the autopilot.
> 
> Any tips would be well appreciated!
> Edit: I see this question has been brought up before! I hope some else has some insight on it!

squilter, 2016-10-28:
There's no good way to gauge distance from the net using only one IR lock beacon. I wonder if it would be better to use 2 beacons instead. I wish I had time to work on this. Seems like a really fun problem.

stormizer, 2017-02-12:
Here to bump up old threads :)
Much awaited.

senosahisnu, 2017-07-13:
Hello! I think the position can be feed from home position, assuming that net position is same as home, that would leave the IR lock for altitude and lateral navigation, much like ILS in manned aircraft.

magicrub, 2017-07-13:
you want to set the NAV_LAND waypoint, not change home. Changing home dynamically has many other side effects that you don't want.

RicoBee, 2017-08-17:
Any luck with the IR Lock integration into the plane world lads?

paocasao, 2020-05-11:
Hi, I want to use precision landing using IRLOCK in a quad plane like in a copter. Could you please tell me if it is possible?