ArduPilot/ardupilot #2909: disable or lock servo gimbal when triggering to prevent jello

R5fan, 2015-09-28:
Im using a cheap servo gimbal to stabilize my camera module, but servo's have limited resolution and move very fast, as a result, even with exposure times of <1/2000th of a second, photo's get jello. Example:

http://i.imgur.com/tqtmU3Gl.jpg

If I disable the servo's or gimbal stabilization, the jello goes away and the resulting motion blur is comparatively tiny because my plane doesnt pitch or roll anywhere as fast as the servo's can. But then of course, my photo's are no longer shot from the correct angle

So please implement a way to automatically disable the gimbal servo's for x ms when the camera is triggered.

magicrub, 2015-09-28:
Sounds like a good candidate for a param. What method do you use to trigger
the photos? Auto mission events?

R5fan, 2015-09-28:
Yes, autonomous survey grids from a fixed wing plane, relay trigger thats read by my raspberry. Here is my latest attempt:
https://photosynth.net/embed.aspx?cid=6144773d-6422-4e74-b9b2-bf7b253313fd

Thats using a 3 gram $25 raspberry camera and $6 worth of gimbal servo's. 
Too bad about the jello, but I think it can be avoided.