ArduPilot/ardupilot #2912: Alarm when landed/landing

ThorstenDP, 2015-09-28:
Hi all,

...it is a longer story...
The essence is that we had to find a copter after a failsafe land in an area with a high density of trees and shrubs. We searched about 1km^2 on the ground an with another survey copter. We were not able to find it although we scanned through 1000 images multiple times.
Luckily, we found the copter 24 hours later the next morning due to the battery failsafe buzzer signal. The battery voltage was down to < 2.5V each cell...
What would be really helpful would be signal when in any land mode (land, RTL, crash detection). I guess some will not like it so one should be able to switch it off. Such a signal would not only help finding a UAV when it landed out of sight but also help to warn anyone who might be in the landing area.

Thanks and best regards,
Thorsten

iskess, 2015-09-28:
Just put a timer on it so the alarm doesn't activate for X amount of time after landing.  Actually, I'd prefer an alarm that activates X amount of time after launch. This way it doesn't matter if you land or crash, the alarm will sound. My copter can't fly longer than :20, so I'll set mine to :30.

magicrub, 2015-09-28:
This is already supported in copter, which can from code I wrote for plane
but I ended up not pushing.

iskess, 2015-09-28:
Great news. I'm always a bit slow on the copter stuff. May I ask why you didn't push it for plane? I see no harm in an option for a simple timer. For my plane I would set it for 1 hour after launch, I should be on the ground by then, and if not, then it sounds while airborne, no problem. The question is how best to silence the alarm if you don't want to reboot? Using the same action as rudder disarm seems logical. However the timer must continue if the plane disarms itself after auto landing.

ThorstenDP, 2015-09-30:
A timer is an option for sure, but it adds operational complexity as well as weight. Hence, I would appreciate a parameter to engage a buzzer sound which continues until the copter is powered off. Maybe the safety switch could be used to silence it.

From a security point of view any failsafe situation should be accompanied by some alarm (like the low battery warning sound) - to warn anybody in the area as well as to be able to find the UAV asap.

iskess, 2015-09-30:
When I say timer, I mean a software timer in the code, not some physical piece of hardware. So no additional weight and complexity. 
I agree that failsafes should have the option to sound the alarm.

ThorstenDP, 2015-09-30:
Ah, ok! But this is nothing which is in AC3.3 or master already - or?

lvale, 2015-10-05:
@ThorstenDP Wouldn't the Lost copter function be enough to locate the copter (both sticks down and right if memory serves me right) ??

ThorstenDP, 2015-10-06:
@lvale 
Ok, I forgot about that! But I guess this cannot be switched on during flight.
Hence, are some situations where this function might not help.
I was not there when it happened so I only have a rough idea what was going on. The pilot was relatively unexperienced. Then the GPS signal was lost, the copter drifted away and the pilot was in panic mode resulting in some "flyway". There are many things you have to be aware of and there are lots of information you can/have to look at when trying to save such a situation. It is a classical BFLE ([bare foot lego event](http://www.suasnews.com/2015/10/38910/suas-risk-mitigation/)). So any additional action required from the pilot should be avoided.
Moreover, I only had two very minor crashes so far when experimenting with new setups. And in both cases the battery came loose. In this specific case no one saw where the copter landed. But I am pretty sure it would have been easier to locate with some warning sound while landing.

ThorstenDP, 2015-10-06:
@lvale 
A sound while landing will also be an additional warning to anyone in the landing zone.

lvale, 2015-10-06:
@ThorstenDP Sure. I suggest that beeping sound trucks do while reversing :) 
like this : https://www.youtube.com/watch?v=eBoM6MrKWwI

:)

ThorstenDP, 2015-10-06:
@lvale The question is: what is the most annoying sound? ;-)

lvale, 2015-10-06:
If you get a PixHawk mini from GLB (I have also regular Pixhawks) sounds won't be an issue :) - No speaker port :).
I tend to  agree that the vehicle should make sounds when "unexpected" events happen, such as the various failsafes, and these events already have their sounds. For regular events like LAND which sound should be produced ? The failsafe event sound or the LAND sound, or both?? Sure there is no failsafe event for "Pilot in panic".

ThorstenDP, 2016-07-27:
Maybe a sound/warning should also be emitted when ADS-B is activated

magicrub, 2016-07-27:
not when ADS-B is activated, but definitely a good idea if an aircraft is nearby