ArduPilot/ardupilot #2915: Ch 7/8 option: clear waypoints

johnbrownlow, 2015-09-29:
The option to map a channel > 1800 to 'add waypoint' is great for creating on-the-fly missions but it would be nice to be able to clear a mission in flight without having to use Tower or similar. So this request would add a 'clear mission' option to Ch7 or 8 so that (for example) one switch could be mapped to 'add waypoint' and one to 'clear mission', so you could restart the whole process.

johnbrownlow, 2015-10-02:
I see that this is a near duplicate of #298

https://github.com/diydrones/ardupilot/issues/298

rmackay9, 2015-10-03:
Thanks for the request.
Personally I think manipulating the waypoints is best done using a ground station like Tower.  It can provides a relatively rich interface including adding, deleting, moving waypoints.  Much more than we can ever do with a limited number of transmitter switches.
This is just my opinion of course, the overall opinion of the users and dev team is how things are decided normally.

johnbrownlow, 2015-10-03:
Unfortunately what you describe is next to impossible if you are flying using a transmitter with an attached monitor which you are using to set up shots. There is just no way to fly the drone, set up the shot on the monitor, then monkey with waypoints on a tablet which is usually sitting on the ground somewhere. 99% if the time when I am shooting music videos solo I am making up the shot on the fly, so there really isn't an option to set it up beforehand, and as I say it's physically impossible to do it using monitor, transmitter and tablet without some kind of rube goldberg monitor-and-tablet rig, which becomes totally impractical if you are hiking into a location or moving around.

R-Lefebvre, 2015-10-03:
Hi John,

I think for the complexity of operation you are suggesting, you really would be better off with a second operator running the ground station, leaving piloting duties to the primary pilot.  There's really only so much one person can do.  For operations hardware setups, I would suggest using a tripod to hold the video monitor and tablet/computer.  That's what I do and it works very well.

johnbrownlow, 2015-10-03:
For my situation, neither of those are feasible unfortunately. I am often in remote hike-in situations and I do not have access to a second operator. Hence my request.

rmackay9, 2015-10-03:
As a side note, I think the Tower guys are looking to extend it so that video can be seen directly within the app.  That could potentially get rid of the need for two screens.

johnbrownlow, 2015-10-03:
That's interesting... thanks for the info. My experience using a tablet as a monitor when shooting (non-drone) video has been pretty gnarly so far, lots of latency and freezing issues. I bought a tablet specifically to control the 5DII but it really didn't work out the way I wanted! Same issues when using the GoPro app in iOS.I think the solution in #298 of 'hold to clear' is a good and simple one, the only downside being lack of feedback that you have actually cleared the mission (it doesn't refresh in Tower... you have to do it manually to see what's changed).

jmachuca77, 2018-07-17:
This functionality has been implemented with cable cam on companion computers. I think that is a much better solution to this. I would recomend that this issue be closed.

jmachuca77, 2018-08-16:
The intended functionality as described a few posts up is the same as the cable cam. I think it would be better to implement something like that on apsync. Maybe port the shot manager from Solo. Openned a new issue on ardupilot/companion#41

closed 2018-08-16 by jmachuca77

peterbarker, 2018-08-17:
While we still have the ability to toggle items in using the transmitter I think it really does make sense to allow clearing it.

It's not even all that hard: https://github.com/ArduPilot/ardupilot/pull/9211

I would really like to unify the behaviour of toggling waypoints in.  The handling of home position is weird, and Copter's "automatically add a takeoff" is even weirder.

rmackay9, 2018-08-17:
@peterbarker, I'm happy for the adding-waypoints auxiliary switch feature to be simplified..