ArduPilot/ardupilot #2920: Arducopter Bug motor PWM values can't be seperated

Jman841, 2015-09-30:
I am bench testing a quadcopter and one motor is a little less powerful then the other 3 so I moved the high PWM value for that 1 motor and kept all the others the same (high PWM of 1800 for 3 motors and the 1 motor is at 1850). 

However, when ever I run the motor test, it uses the highest value for all motors, not just the one. I verified this by looking at the PWM output in the status page of mission planner. I can't seem to unlink the motors. It always takes the values from the highest of the four motors.

R-Lefebvre, 2015-09-30:
Can you show your mods?  Hard to comment otherwise.

Jman841, 2015-09-30:
What do you mean by mods?

If I set 

RC1_Max 1800
RC1_MIN 1100
RC1_TRIM 1450
RC2_Max 1800
RC2_MIN 1100
RC2_TRIM 1500
RC3_Max 1900
RC3_MIN 1100
RC3_TRIM 1450
RC4_Max 1800
RC4_MIN 1100
RC4_TRIM 1450

The pixhawk will put out a mid value of 1500 and a max PWM value of 1900 for ALL motors instead of only for RC3 which is what I want.

Jman841, 2015-09-30:
I tested this using the Motor test, and also simulated flying with it strapped down to a bench tester. Same result either way, the pixhawk will output a max value of 1900 PWM for all 4 motors instead of only RC3 which should be.

R-Lefebvre, 2015-09-30:
Ah, OK.

So the issue is that RC3, and only RC3, is very specifically the "master" throttle output channel.  If you did the same to RC1, 2 or 4, you would have no change in throttle behaviour at all.

Jman841, 2015-09-30:
haha of course, the only one I need to adjust. Is it possible for this to be fixed?

R-Lefebvre, 2015-09-30:
I don't think so.  Certainly not easy.  And I'm not sure you should do that anyway.  Have you calibrated the ESC's?  Maybe the issue is that one has a different calibration? 

If it really is just the motor or prop, it will be dynamically adjusted by the Rate PIDs quite well.  Every seen Marco's water bottle flying videos?

To try to do what you ask would be a very substantial change to the code.

Jman841, 2015-09-30:
Ah ok, No. the ESC's are KDE ESC's so they used fixed endpoints for PWM values. One motor had some slight damage to the wire and does not get to full power. The other 3 motors have ESC sync issues on the bench above 1800 pwm with the current voltage and setup so I limited them all as there is still plenty of power. The one motor (RC3) does not have sync issues at 1900 pwm so I was pushing that one to the full throttle range to compensate for the bit lower power. 

Its been an interesting time trying to deal with sync issues on this setup and I think we are almost there with the latest KDE Firmware. But, it should fly fine with all of them at 1800 pwm.

R-Lefebvre, 2015-09-30:
Really?  That's very interesting.  I thought KDE blamed the Sync issues on lack of 5V power supply to the Opto ESC's?  That never made sense to me.  I suspect they're not really opto.  We should talk elsewhere about your experience with these as I was about to use them on a new helicopter, but I might stick with Hobbywing.

Jman841, 2015-09-30:
Shoot me an email and i'll let you know what our tests have shown so far.

expntly, 2018-05-10:
@Jman841 @R-Lefebvre 

Sorry for resurrecting this issue almost 3 years later but we're tracking down KDE ESC sync issues and searching github.com/ArduPilot/ardupilot for KDE related info.  This particular issue appears to be one of the only two results.

Have you  figured out the sync issues and what/where they originate from?  Thanks!

raygamero, 2019-02-05:
@expntly What type of sync issues are you having.
I'm having a problem getting a KDE ESC/Motor pair to start properly.

rmackay9, 2019-02-06:
I've changed the title a bit to better reflect the request.  There is currently no way to allow separate PWM ranges for some motors.

In this particular case it seems to be a hardware issue (there is talk above of "damage to a wire") and I'm not sure we should be attempting to fix hardware issues in software.  Knowingly using damaged ESCs or Motors on a multicopter is a recipe for disaster.

I think I might close this issue actually because I haven't heard of anyone else requesting this and I think adding this flexibility to individual output ranges is more likely to **cause** setup issues for users rather than cure them.  Happy to be corrected if other disagree.

closed 2019-02-06 by rmackay9