ArduPilot/ardupilot #2921: Plane: inhibit switching to land_approach stage until additional criteria is met

magicrub, 2015-09-30:
When reaching the usual land approach waypoint, inhibit switching to FLIGHT_LAND_APPROACH until certain criteria is met:

- next waypoint is LAND (already implemented)
- target_bearing is +/- 10 deg of actual_bearing
- have progressed forward along path by at least 5%

magicrub, 2015-10-01:
also current_loc.alt < prev_WP.alt

magicrub, 2015-10-01:
I won't push this as a PR yet, but here's a commit that does it:
https://github.com/Airphrame/ardupilot/commit/ab0a1a78b7f25267806bf4bfe965123b52a09cd1

Tridge, over mumble, expressed not using location_path_proportion() because that doesn't take into account waypoint radius which can skew those numbers quite a bit on short approaches. He suggested just making sure you're going the right way via this pseudo code:
making_progress = (angle_between(vector_from_vehicle_to_landing_pt, ground_course_vector) < 90degrees)
if (making_progress && (path_proportion > 0.3 || lined_up)) 

But I can't get my head wrapped around it enough to write actual code to do it.

iskess, 2015-10-01:
Thanks to you, I'm starting to understand why I get such irratic behavior on the turn to final approach during an autoland. As a work around until this code is done, could we simply add another waypoint just outside the WP radius after the hard turn? That should prevent the switch to the final approach phase from being co-located with a steep 90 deg turn.

magicrub, 2015-10-01:
Thanks @iskess i like digging into this stuff. 

But no, the autopilot will not be generating extra waypoints to smooth things out. As the mission planner, you can :) just plan your mission with it like that. I've recently been using a short level plateau before the approach so when I switch from normal to approach I'm already going straight.

iskess, 2015-10-01:
Thanks Tom, that's all I meant by a work around. Just manually adding an extra WP in the flight plan. I haven't tried it yet, but I'm hopeful it will help my approaches while you work on the code for 3.5. 

What are the conditions that trigger the switch to Approach Mode? Is it only crossing the last WP?

magicrub, 2015-10-01:
Right now, yes. Technically it's whenever you are navigating to a LAND
waypoint.

magicrub, 2015-10-21:
pending PR for this: https://github.com/diydrones/ardupilot/pull/3013 I've flown it and it looks great. Only minor suggestion I have is to loosen the restriction of needing to travel forward from 30% to 20%.

magicrub, 2015-11-11:
PR accepted, closing issue

closed 2015-11-11 by magicrub