ArduPilot/ardupilot #2923: Multiple battery monitoring - combine amperage

DaveMartin, 2015-10-01:
APM 3.3 introduced BATT2 as a totally independent monitor; would request now adding an option to sum BATT and BATT2 current for consumption monitoring and COMPASS_MOT.

Two independent voltage monitors is fine for me, one to monitor the flight power source and one to monitor the avionics battery. I'm not really worried about monitoring the avionics current, but I would like to combine the BATT and BATT2 amperage to help monitor the flight power, both for consumption and for COMPASS_MOT

My flight power is two packs, each (for resilience) wired to the power distribution board. I appreciate if I used a 'Y' lead I could use one ATTOpilot sensor; but I have two separate leads, each with their own in-line sensor, soldered to the power board, one per pack.

As the packs as paralleled up, the voltage will be common. The separate currents, if logged/monitored, may give an extra indication if one battery is mismatched or running down faster than the other, so two independent current/mAH figures are (slightly) useful.

The reason for the request is that, for those who run parallel batteries in maximum resilience configuration, unless the two amperages can be summed in real time, COMPASS_MOT cannot be used.

Dave Martin

magicrub, 2015-10-01:
I think BATT2 is actually for pixhawk 2 where there are two power sources. Is that right Phillip? @proficnc

Naterater, 2019-01-08:
This has been implemented as option 10 of BATT_MONITOR

http://ardupilot.org/plane/docs/parameters.html#batt-monitor-battery-monitoring

closed 2019-01-08 by Naterater