ArduPilot/ardupilot #2928: Plane: add AOA (Angle of Attack) sensor

magicrub, 2015-10-01:
(moved from DD thread by Jason Franciosa)
If we could develop a cost effective AOA sensor that is reliable I think this is the ultimate solution for stall prevention. Although, I don't think it completely rules out the need for stall detection and recovery as nothing is 100% bullet proof

magicrub, 2015-10-01:
Continuation of DD thread: This is a continuation of https://groups.google.com/forum/#!topic/drones-discuss/HBDzMB5WLa0

We still need to keep on the TODO list:
1) bullet-proof stall prevention (add AOA)
2) but when that fails detect the stall
3) once detected, try to recover https://github.com/diydrones/ardupilot/issues/2669

It's important to implement them all but to focus on the stall prevention which can be improved with AOA.

Jman841, 2015-10-01:
Sounds about right. The biggest issue at hand right now is lack of the AOA sensor. As brought up in the DD thread there is a few DIY ones that seem pretty basic.

Correct me if i'm wrong, but, the code part shouldn't be to difficult as it will basically use the current stall prevention code that is reliant on the airspeed sensor and swap it with the AOA sensor, with new values of course.

proficnc, 2015-10-01:
[image]
there are a few ways to detect AOA (And side slip) 
but the simplest for us to test, would be a differential pressure sensor between two tubes, that have been cut as per the picture.

the tubes can just be simple copper for the test, soldered together.

using the old analogue Airspeed sensor, we can feed this into an analogue pin, and log the results.

kwikius, 2015-10-02:
@proficnc . +1 Much prefer this solid state solution to a weathervane approach. Much more robust. Not sure that it needs to be 3D though. Wouldnt 2 tubes  ( one up, one down) suffice initially?

regards
Andy

R-Lefebvre, 2015-10-02:
I would also be interested in these devices as an air direction measurement for helicopters.

geofrancis, 2015-10-02:
To detect a stall,  check if if the plane is nose down with the elevator hard up for more than 2 seconds and doing a complete yaw rotation,

 to pull out the stall it would just need to switch to mode 1 acro mode for 1 second then back to flybywire/auto, that should give it enough airspeed to recover.

by switching to acro briefly we stop the plane trying to level and just get it to kill its rotation this will drop the nose and allow the plane to pick up enough airspeed that it can pull up.

magicrub, 2015-10-02:
Hi @geofrancis 
This is the AOA thread. For stall detection and recovery please visit https://github.com/diydrones/ardupilot/issues/2933 and https://github.com/diydrones/ardupilot/issues/2669.

phollands, 2015-10-04:
Note: For some purposes, AoA can be estimated using the IMU data  without recourse to air sensors. This was part of the theory and practice behind the development of the control logic in another autopilot project helicalTurns ( Bill Premerlani). Just thought you should keep this in mind, although measuring actual angle of attack with sensors  may be better when close to the stall. Our goal was somewhat different.... simply executing accurate steep turns in fly by wire mode.
[image]

Georacer, 2015-10-10:
Hello eveyrone,

I'm writing this post on behalf of the Basic Air Data project, which I am part of. We are only 3 core members, but I believe we do have some background to contribute to the discussion.
This project is dedicated to the design and manufacturing of air data instruments, so it fall right in the middle of this discussion.

First and foremost, let me show you some of the fruits of our labour, to set a baseline for the discussion:
* A better, more accurate (albeit more expensive) [Pitot probe](http://www.basicairdata.eu/pitot-tube.html) which has also been manufactured successfuly
* A full

Georacer, 2015-10-10:
Oh, btw, in case it means anything, the two of us have DIYDrones profiles as well:
* http://diydrones.com/profile/Georacer
* http://diydrones.com/profile/JoseLJr

Sometimes too many platforms break internet continuity. ;)

marcmerlin, 2016-02-21:
AOAs have been missing on full size planes for eons, and it's a damn shame.There are 3rd party kits like this one, and they're awesome for safety: http://www.alphasystemsaoa.com/

I couldn't find very good pictures of the sensor, but here's one: http://www.alphasystemsaoa.com/duplicate-of-lancair-es.html

magicrub, 2016-02-21:
Looks like a pitot tube at an angle. Not very rocket-sciency! hehe

WickedShell, 2016-02-21:
Heh, I've gotten quotes before for pitot tubes that have additional holes
and grant AoA, and sideslip as well as airspeed (and a higher range of
calibrated incidence angles then the tubes we use now). Was really cool,
but the price tag/weight was still to high, I'm looking forward to when
these drop down in cost/weight.

iskess, 2016-02-21:
Maybe it could be made as a solid rectangular bar like the SenseFly ebee. 3D print the rectangle with channels to fit thin tubes at different angles.

kwikius, 2016-02-22:
I have seen angle of attack sensors which consist of 2 plate surfaces at nominal + -45 degrees to the air flow. Each plate has a hole central where the pressure differential is read. 

Another wild idea. Use an array of LDR or photodiode covered by casette tape along wing just behind 30 perecent back from L.E. The casette tape will lift as stall approaches thus exposing sensor to light :) Same principle as tell-tale. http://www.wyc.org.uk/fevanationals/docs/luff_telltales.pdf

regards
Andy

EShamaev, 2016-12-14:
Why is separate sensor is needed? We have all sensors already.
https://www.aspenavionics.com/aoa
https://www.aspenavionics.com/news/aspen-introduces-integrated-angle-of-attack-aoa-indicator
http://www.flyingmag.com/avionics-gear/instrumentaccessories/we-fly-aspen-avionics-aoa-indicator

magicrub, 2016-12-14:
From an article you linked:
 "The three primary inputs the system uses to determine AOA are acceleration, airspeed and pitch."

As your pitch angle increases, the airspeed sensor gives incorrrct values.

EShamaev, 2016-12-14:
It can be compensated, however in +/- 15 degrees range that is usefull for us, most airspeed sensors have realtively small error.
The main article I wanted to base on is this:
http://folk.ntnu.no/torarnj/aoa_ssa.pdf
with some ideas from Aspen's implementation.

marcmerlin, 2016-12-14:
More generally, an airplane will stall at a different airspeed depending on its wing loading (i.e. weight). I've had a TB20 give me a stall warning at 90kts while taking off from Reno on a hot day, fully loaded (i.e. thin air, high wing loading) when that plane's approach speed is 73kts and it's not supposed to stall above 65kts
Point being that angle of attack and how far away you are from stall cannot be well determined with your airspeed. For precise "how far am I from stall", you do need an AOA sensor.
That's what airliners and the airforce/navy use for landings, so they are probably right :)

EShamaev, 2016-12-14:
The AOA sensor if we are taking the commercial aircrafts are mostly like this:
http://www.ameteksensors.com/Products/Angle-of-Attack-Sensors.aspx
[image]
[image]
[image]
It merely measures the angle of air passing to the centerline.
How does the AOA sensor knows about the wing load?
The stall prediction is really made by computer with estimated current weight, airspeed, AHRS data, so on. But this is only minor corrections and **unlike airspeed the wing always stalls at the same critical angle of attack**. Please do not mistake steep descent with stall, these are completely different things.
Critical angle of attack (stall angle of attack) does not depend on the load, the profile of wing does not change with mass of plane.

Also in UAV (especially battery powered) the mass does not change in such wide range.

marcmerlin, 2016-12-14:
1) my mass can change quite a bit (by 20-30% when I add new cameras or put long range batteries)
2) stall is indeed always at the same AOA, and higher wing loading forces you to use a higher AOA, causing you to reach that critical AOA more quickly. An AOA sensor should not care about your weight.

EShamaev, 2016-12-14:
1. I agree, in my plane mass changes about 40-45%, I have 12 liters of fuel which is burned out during flight.
2. Exactly. That's my point.

Also regarding @magicrub "As your pitch angle increases, the airspeed sensor gives incorrrct values."
It does not. It gives incorrect values at changes of AOA. You can have 0 AOA with air passing by the pitot perfeclty straight with positive or negative pitch.

I plan to add the calculation of AOA from values of velocity vector and wind vector with some slight need for air speed (http://folk.ntnu.no/torarnj/aoa_ssa.pdf) and add the corresponding indictor in mission planner.
Will this be welcomed by community?

Jman841, 2016-12-14:
Any development on the hardware or software side for an AoA sensor?

EShamaev, 2016-12-14:
Only software. We already have all hardware.

marcmerlin, 2016-12-14:
@EShamaev I think @magicrub is correct. If you are travelling in level flight but you're pitched up 15 degrees to generate enough lift to stay at your altitude, your pitot tube is angled 15 degrees compared to the relative wind and therefore does not get the same airflow going in and will show a lower airspeed than what's actual. So, in slow flight your measured airspeed becomes more and more incorrect.
I've verified this when practising slow flight in full size aircraft and flown them at 40kts indicated (full power nose up at the edge of stall), and gotten airspeeds that were slower than the speed I should have stalled at.

EShamaev, 2016-12-14:
I agree but my point was that airspeed error is not from pitch, it's from AOA.
However this can be iteractively corrected by each other. I will investigate stability of such correction, but from first point of view that is linear and negatively feeded back to each other and should be stable.

EShamaev, 2016-12-14:
By the way, @marcmerlin, on you full size aircraft, do you have static ports on pitot itself or on fuselage?

iskess, 2016-12-14:
Wouldn't the airspeed error caused by the increased pitch angle be predictable and part of the equation? This, together with the accelerometer's output of the load factor should be enough data to substitute for an AoA sensor, in theory. 
Weight changes are still an issue. Maybe there could be a GUI on the GCS that allows for adjustments for different battery and payload configuration.

EShamaev, 2016-12-14:
Weight changes are not an issue. It has nothing to do with AOA.
Airspeed error due to change in AOA can be propagated to AOA and AOA can be propogated to Airspeed error correction. Both terms seems to be negative, so the corrections should be stable.

EShamaev, 2016-12-14:
Have found following:
http://www.unitedsensorcorp.com/pitot-properties.html
_probe yields velocity and weight flow readings accurate to 2% up to angles of attack of 30°_

Other references:
http://www.tandfonline.com/doi/pdf/10.1080/19942060.2010.11015340
http://spaceagecontrol.com/nasa-tm-104316.pdf
https://www.nasa.gov/centers/dryden/pdf/87961main_H-1189.pdf
https://www.nasa.gov/centers/dryden/pdf/87803main_H-665.pdf
Give roughly same numbers for loch Mach numbers.

magicrub, 2016-12-14:
Eugene,

You are correct, it's the angle of the wind that is critical and not the
angle of the aircraft. A common assumption is wind is 0deg. That is correct
most of the time but the purpose of the AOA sensor is to confirm/measure
that.

On a side note, I'd like to mention that Ardupilot does not store a 3D wind
vector, only 2D. It is computed in the EKF matrix as a side benefit but
then discarded because it has shown to be very noisy and misleading.
Especially for copters.

EShamaev, 2016-12-14:
Yes, I have found this:
_wind.z = 0.0f; // currently don't estimate this_

Well, anyway I will be implementing AOA and SSA readings with link to airspeed, lift factor and mass for my purposes (stall prevention). Even based on the assumption of zero vertical speed this should give reasonable values to start with.

The only question I have now, is where it should be done.
I think of implementing members to EKF2 class getAOA and getSSA.
But then apart from using them in the plane, how it would be better to send them to GCS?

magicrub, 2016-12-14:
Put it In the ahrs where the wind is.

EShamaev, 2016-12-14:
Allright, will go to AP_AHRS top level class.
How is it better to send it to GCS?

magicrub, 2016-12-14:
Is there an existing MAVLink packet for it? If not then a new one should be
made and brought into mission planner.

Don't forget to log the value to data flash!

EShamaev, 2016-12-14:
Logs will be the first thing to do.

I have not added any packets to MAVlink before, please advice if this:
http://ardupilot.org/dev/docs/code-overview-adding-a-new-mavlink-message.html
is actual way to go?

magicrub, 2016-12-14:
> mavlink-message.html

Yup, looks good except The end where it mentions dinner discuss. That
should be changed to say gitter.im/Ardupilot/Ardupilot

marcmerlin, 2016-12-15:
@EShamaev all full size planes I've flown have the static port on the side of the aircraft, usually behind the passenger doors. There are multiple of them because you wouldn't want a single one being blocked causing loss of altimeter. Actually there has been at least one airliner crash because the static ports had been taped by cleaning crew. 
As you know, losing static ports also tricks your airspeed into rising as you go up, causing the avionics to think you are in an overspeed situation, then it pitches up, causing your altitude to go up, perceived airspeed to increase, so it pitches up more until you stall.

EShamaev, 2016-12-15:
Yes, that's why I asked and that seems to be an answer to much higher error at high AOA. From studies above - the best behaves the Prandtl pitot tube and it has least error with change of AOA as the static port balance the difference. When you put static ports on fuselage, they do not provide as much compensation and you get unrealistic readings at high AOA.
If we consider pixhawk digital sensor pitot tube that is exactly Prandtl tube, we should not be getting more than several percents of error over reasonable AOA range.

marcmerlin, 2016-12-15:
You make a good point, honestly I have no idea what reasons there are for putting the static ports so far away outside of reliability and redundancy issues. I suppose they may want to guard against the pitot tube being blocked and in that case you still have altitude and GPS speed, whereas if static port were on the pitot tube, you'd lose both? Just guessing, maybe the real answer is that there is no good reason outside of it's always been done that way.

EShamaev, 2016-12-15:
Well, actually on planes that operate near or over critical AOA (fighters, aerobatic), usually some modified Prandtl tube is used as shown in above US gov and NASA research papers. But such designs are not easily manufactured in our scales, so for us best would be to go with standard Prandtl tube.
As for big airliners, as they have separate sensors for AOA, the pitot values might be corrected by air data computer with input from direct measure of AOA and SSA. I can not calim this, as I have not studied that specific area, but seems to be reasonable guess.

EShamaev, 2016-12-15:
Some more info regarding pitot tubes and sensors for UAVs. I have found manufacturer who produce small AOA and SSA sensors as well as very nice Prandtl tubes for UAVs.

http://www.swiss-airdata.com/
http://www.swiss-airdata.com/smv1

EShamaev, 2016-12-16:
https://github.com/ArduPilot/MissionPlanner/pull/1433

https://github.com/ArduPilot/ardupilot/pull/5400

Jman841, 2016-12-17:
@EShamaev What is the existing AoA hardware you are using or are developing for? Is there any low cost solutions that are readily available or easily created?

EShamaev, 2016-12-17:
@Jman841 Not sure what you mean. I am implementing an estimation of AOA and SSA in software without using specific new hardware.

Jman841, 2016-12-17:
Ah, sorry for the confusion, I didn't realize that. Very cool! Thank you for putting in the hard work on getting this developed!

geofrancis, 2016-12-20:
why dont you impliment a general basic input for a AOA sensor and a stall sensor.

AOA just needs an analog input with a potentiometer that can be calibrated and stall detection could be tripped if it goes over a set angle.

a stall sensor just needs to be a digital input, so if pin=high then its stalled.

then there is no need to debate about hardware, people can go out and make their own from whatever as long as it conforms to those basic specs.

EShamaev, 2016-12-20:
It's not that simple. Potentiometer will not work as the sensor needs to have extremly low friction to measure correctly the AOA and even by deflected by wind at all at our speeds. Above I have posted a link to company that makes such sensors for UAVs. Price is very high.
Stall sensor is a whole different story. You can of course install a small deflector with microswitch that will detect some of the possibilities of separated flow, but it will not work in all stall conditions and may trigger in some conditions when actually not at stall.
In overal, I agree that sensor based solution for AOA and SSA is the best, but for majority of users, the price of sensors that will really work is way too high.

magicrub, 2016-12-20:
Sorry geofrancis, it's not that ready. A Stall is basically when (lift <
weight). Lift is a function of airspeed over the wings, AOA is effective
airspeed over the wings, and the wing loading is the effective weight.

EShamaev, 2016-12-20:
I completely disagree.
Stall is not when (lift < weight). Stall is when airflow separatates from top wing surface and lift dramaticly decreases even if the wing is at high angle of attack. (lift < weight) is absolutely normal condition that happens when you lose altitude (intentionally or not is second question), but it's not a stall.
Secondly in stall the aileron function to control roll is greatly decreased and instead they are taking part in yaw action which leads to large yaw moment. This leads to a most known aftereffect of stall - uncontrolled spin. But spin itself is not a stall, it's what happens next after stall in most cases.
Also AOA is not effective airspeed, it's and angle of attack of the wing against the airflow.

geofrancis, 2016-12-20:
i just think its a bit if a chicken and egg problem, if there is no software support then they wont get made.

it doesnt have to be a potentiometer specifically but just something that will do 0-5v for 0-360 or values that can be calibrated.

if the support is there for a stall sensor then people can make thier own choices about how they make it and how much use it is

EShamaev, 2016-12-20:
There is support - code is open. My point is that people who can make such sensor will definetly be able to integrate it in code.

geofrancis, 2016-12-20:
its really not hard to find very low friction resistive rotary sensors 

http://www.alps.com/prod/info/E/HTML/Sensor/Position/RDC50/RDC50_list.html

EShamaev, 2016-12-20:
It's extremely easy to send ADC parameter to GCS already.

On pixhawk:
Virtual Pin 13 and ADC 3.3V connector Pin 4: This takes a max of 3.3V. May be used for sonar or other analog sensors.
Virtual Pin 14 and ADC 3.3V connector Pin 2: This takes a max of 3.3V. May be used for second sonar or other analog sensor.
Virtual Pin 15 and ADC 6.6V connector Pin 2: analog airspeed sensor port. This has 2:1 scaling builtin, so can take up to 6.6v analog inputs. Usually used for analog airspeed, but may be used for analog sonar or other analog sensors.

Easiest is to treat it as an analog sonar input but not used in control loops. You can set it up and get displayed in GCS.

Also transmittion of a custom value to GCS is also very simple to code.

magicrub, 2017-03-20:
Here's a nice video from Paparazzi devs, it shows a nice paid of AoA sensors: https://youtu.be/mlm0aW8NubA?t=73

EShamaev, 2017-03-20:
Yes, very nice! However from video you can see that it does not work great, though these were very easy to implement.

Georacer, 2017-03-24:
We had recently done some wind tunnel calibration of airspeed and AoA sensors.
We decided to write a report on our findings and I just posted it in http://discuss.ardupilot.org/t/low-cost-air-data-instruments-calibration-and-discussion/16024

I think it is relevant to this discussion.

EShamaev, 2017-03-24:
Very nice write up!

OXINARF, 2017-06-04:
AOA estimation has now been merged in #5933, do we still need this issue open?

magicrub, 2017-06-04:
I think this can be closed. Any additional work would be additional and would be best done under a new issue. Closing...

closed 2017-06-04 by magicrub