ArduPilot/ardupilot #2931: Trigger a parachute on long failsafe action FS_LONG_ACTN

severinleuenberger, 2015-10-02:
On the view of the UAV-Operator and owner it is most always the best scenario to have RTL/RTH for rescuing the airplane. 
Regarding official governmental rules, we face other needs. While discussig with government authorities, we heard, they want to have the UAV landed securely by parachute on signal lost. Nothing else than a paracute. 
We need to use the long failsafe action FS_LONG_ACTN, to trigger aparachute by a signal who changes from PWM 1000 to 2000 on one of the chanels AUX1 to 6. 

Tho chose on what chanel the Trigger should work, I suggest to implement a new Parameter
FS_LONG_AUX

We would appreciate very much, if APM could implement this funcion. 

Best wishes
Severin / Switzerland

NiqueIt, 2015-10-13:
I fully support the request and can add my experience with gov.

The need comes from the approval to fly beyond line of sight. If you lost the plane/copter, the aircraft is not a risk to itself and if only the link is gone, it can return home - technically. But meantime, it poses the risk of midair collision as it can not be seen by other aircraft - and for sure, the drone can also not react on a collision course.

So RTH is good to safe money and have the drone back, but it is not a suitable solution to get the permission to fly BVLOS.

To trigger a parachute is a better solution, there may exists even more. But anyway, they need to be triggered in any way - but most via a pwm signal on an output. Why? Because any ERC (Emergency Recovery Capability) should be independent from the FC (Flight Computer). So you need a hard-wire between the two systems.

severinleuenberger, 2015-11-28:
Any news on this? If you need any help as such as informations, I can definitely help. But I cannot implement it my self (lack of knowledge).

squilter, 2016-01-19:
fixed

closed 2016-01-19 by squilter