ArduPilot/ardupilot #2933: Plane: stall detection

magicrub, 2015-10-02:
Stall detection is needed. Must work with, and without, pitot tube.

related to stall recovery issue https://github.com/diydrones/ardupilot/issues/2669

magicrub, 2015-10-02:
There are many indirect ways to declare our state to be "stalled". My favorite candidate is roll and roll_demand diverge but for this feature to be robust we need to define what a stall is.

marcmerlin, 2015-10-12:
Since I happen to be an airplane pilot, you are correct that you can get one wing that stalls first and that will give you a clue that you stalled as the wing dips.
However, you can have a coordinated stall where no wing dips and just the nose stalls.
IMO, it's easier and more reliable to notice that the nose just dropped a bunch without any control input asking for it to do so.
(actually, "dropped" is a misnomer, you can get an accelerated stall in a 60 degree bank angle where the nose buffets and basically stalls but not towards the ground. You have compute the direction vector for your current attitude and see that the nose slipped from there).
Your other stall is a stall/spin where indeed one wing drops and depending on the plane, you can end up in a spin.
https://www.youtube.com/watch?v=F6d4VKdc34A (watch 10 seconds after start)

magicrub, 2015-10-12:
Thanks for the insight Marc. I think when we are actively in a stall the aircraft is basically not responding to controls so pitch, roll and yaw will all diverge - not just roll out pitch. The question is what will diverge first and by how much and is that a reliable source for a trigger or do other things need to be considered? We need as much of a generic solution as possible.

If we can't get something totally generic, we could look at specific airframes via params such as if the are elevons configured. If we go that route, we would need a table of what aircraft shows what signs of a stall. That may get complicated but would hopefully be more reliable.

In all cases I think it's safe to say our commanded altitude and desired altitude diverge.

marcmerlin, 2015-10-12:
Agreed, going with commanded altitude vs actual altitude is a pretty safe way to detect almost all stalls (except accelerated buffet stalls in bank turns, but those are likely pretty rare in the list of stalls people see, and not something I'd be worried about anyway).

Either way you have a +1000 from me on stall detection and recovery :)

magicrub, 2015-10-12:
Does that mean you're volunteering to flight test? :)

marcmerlin, 2015-10-12:
On Mon, Oct 12, 2015 at 03:51:07PM -0700, Tom Pittenger wrote:
> Does that mean you're volunteering to flight test? :)

While I'd be sad to crash my $1000+ long range FPV glider, I can probably
try this at altitude if the code isn't going to do reboot the pixhawk or do
something too extreme I can't manually recover from :)

Incidently I just tried stall prevention in AC3.4 this morning in training
mode. I'm not really sure it worked, I was able to pull up the nose a lot
and get the aircraft to stall. That said, it may just need more tuning.

Marc
-- 
"A mouse is a device used to point at the xterm you want to type in" - A.S.R.
Microsoft is to operating systems ....
                                      .... what McDonalds is to gourmet cooking
Home page: http://marc.merlins.org/

iskess, 2015-10-12:
I disagree that accelerated stalls are rare, I think they are very common and the most dangerous, especially the turn from base to final approach close to the ground. 
Every unexpected stall I experienced on an RC plane was related to an aggressive roll. That downward aileron stalls the wing you are wanting to lift, and often leads to a spin. I believe divergence in roll will be the most important indicator to the most dangerous type of stall. The ailerons should immediately go neutral and the rudder and elevator need to be used to recover. 
I volunteer for flight testing. Sounds like fun!

magicrub, 2015-10-12:
Ha! Thanks for the enthusiasm Iam. Glad to hear you're on board for
testing. You're always up for a challenge :)

Let's try and keep the stall recovery chat to the stall recovery issue.
It's easy to get side tracked on that topic because I'm sure it will get
complicated for various aircraft types.

magicrub, 2016-02-07:
Ok guys, I may be doing some stall testing next week on my skywalker x8 flying wing. They're notoriously hard to recover from so recovery is a secondary goal. Lets start posting some logs of stalls of different aircraft and situations so we can get a model of what a stall looks like data-wise. Here's a couple things to look for in the logs:
  - [ ] abs(altitude_demand - altitude) is large
  - [ ] abs(roll/pitch/yaw - roll/pitch/yaw_demand) is large
  - [ ] rather large unexpected sink rate.
  - [ ] (airspeed_demand - airspeed) spikes up then all of the above happens.

Anything else we can look for?

iskess, 2016-02-07:
Here is a nice example of a tip stall during and autolanding in a 2013 Skywalker 1880 a few years back. It includes a video for the full experience.
http://diydrones.com/forum/topics/autolanding-failure-tip-stall-on-turn-to-final-causing

iskess, 2016-02-07:
Here is a stall/spin/crash during the autotuning of a Multistar Twinstar II. I believe it was my correspondence with Tridge and Paul over this crash that lead to the STAB_PITCH_DOWN parameter. I didn't realize that FBWA would fight the plane's natural nose down tendency when the throttle is cut, and allow the airspeed to bleed off. Combine that with the steep bank angles required for auto tune, and it was a recipe for a spin. 

[2014-04-29 15-11-20.zip](https://github.com/diydrones/ardupilot/files/120562/2014-04-29.15-11-20.zip)
[image]

magicrub, 2016-02-07:
Thanks Iam for the log. It's pretty old and doesn't have much data in it
but I do see that the gyro shows signs of the stall. Makes sense, when you
stall you'll typically spin. Yet another marker we can scrutinize.

magicrub, 2016-02-07:
Looking at some of my own logs I see the pitch and roll diverged at about the same time, but not by much at first. Pitch slowly starts to diverge first then roll, however the pitch is easier to spot because it clips at LIM_PITCH_MAX which was 20 in my case.

- altitude demand is not logged, but it doesn't drop much until later
- roll/pitch - roll/pitch_demand is indeed large. Pitch shows divergence first but not by much, just a constant diff value for 5 seconds where it's struggling to maintain pitch which is causing the airspeed to drop. Then the roll - roll_demand spike as the wing drops and the death spiral begins at which point the pitch becomes obviously bad too.
- sink rate went from 0 to 20 m/s but that took several seconds to build up. Late indicator of trouble.
- airspeed (10m/s) dropped to below ARSPD_FBW_MIN (12) as the pitch was struggling and stayed low when the roll diverge happened. Once that happened it grew to 20. Looks like my ARSPD_FBW_MIN is set correctly at 20% above actual stall speed! :)
- the gyros all went nuts just after roll diverged.

of note, this log was a stall during a u-turn starting the landing approach which is what prompted https://github.com/diydrones/ardupilot/issues/2921

conclusions:
- an integrated pitch error while airspeed is below ARSPD_FBW_MIN may be the best early detector where pitch and roll demand are good eventually.
- need more _demand value slogged
- sink rate and high x-gyro is a clear indicator that you're falling and spinning out of control
- pitch and roll fail at about the same time but pitch clips to LIM_PITCH_MAX just before roll diverges
- ground speed stayed up so that's not a good indicator ( I was thinking sink vs ground speed should show a vertical fall but it doesn't)

ragflyer, 2016-02-13:
Hi, I'm just beginning to dive into the technical side that deep, but can we get any useful information from the gyros "going all nuts"? Ultimately we should be able to detect spin direction and counter rudder to stop it faster before applying pitch and then throttle (on traditional airframes).
I'd be happy to test code as well (Cessna-like model ready, flying wing follows sometime this summer).

magicrub, 2016-02-13:
The primitive behavior now is to pitch down and throttle up if the airspeed
is too low. It's a 2 stage approach to do anything different: detect an
actual stall and then do a specific action to recover depending on the
aircraft type.

There is another issue regarding how to recover, this issue is for
detection.

I plan to make a stall confidence scheme like is-flying. Gyros going nuts
is a dead ringer for a stall but there are many other signs that happen
before that. Read this issue thread for more info.

If you have some stall logs you can post for analysis please do, more info
the better!

Jman841, 2016-04-14:
I just had a tip stall if the logs can help at all I can post them.

magicrub, 2016-04-14:
yes please, post a link here. When I/someone gets around to working on it then we'll have some data to crunch

Jman841, 2016-04-15:
Here's the Logs:

https://www.dropbox.com/s/tzsmfcwcf3zvdy9/7%204-14-2016%2010-31-28%20AM.bin?dl=0
https://www.dropbox.com/s/w8seipoxwee3eel/2016-04-14%2010-13-56.tlog?dl=0

Jman841, 2016-09-21:
Any updates on this? IMHO, the way Arduplane reacts in a stall is the most fatal component of the system at the moment. Stall detection and a better attempt to recover from it is critical, even something as simple as center ailerons until an angle is established, apply up elevator, and possibly apply power if it is above a certain altitude?

magicrub, 2016-09-21:
Reaction is ready, it's a reliable detection that is the important hard
bit. At what point does the flight controller simply give up and no longer
attempt to fly the plane and declare it's in a stall? False detections are
fatal as well.

Jman841, 2016-09-21:
If it is pointed at an angle greater than 80 degrees down in any auto mode can it be assumed to be stalled? The controller seems to always know if the nose is pointed down, and I don't know of any auto modes that use a 80+ degree dive, possible reverse thrust landings?  But maybe during landing with reverse thrust is on the stall detection is disabled?

200Volts, 2017-01-19:
If ailerons ALWAYS go full tilt left or right in a stall, then make a aileron timer or loop counter to detect a stall.  So, if aileron is full deflection for more than 1 second, set ailerons to neutral, full opposite rudder, full down elevator - until gyros say spin is stopped (i.e. gyro g less than X left or less than X right).  Full recovery would then be throttle to idle, elevator going up until plane is level, at the right speed.

Naterater, 2019-01-04:
> The primitive behavior now is to pitch down and throttle up if the airspeed
> is too low.

Unfortunately this behavior is causing major problems when an airspeed sensor fails.
https://github.com/ArduPilot/ardupilot/issues/9372