ArduPilot/ardupilot #2937: Getting "compass variance" error during flight

msk7-ripe, 2015-10-04:
While testing v3.3 I'm getting compass variance error. Compasses were calibrated pre-flight.
Here is log file http://t13.mediamax.pro/APM/31.BIN See auto mission at the end of file.

rmackay9, 2015-10-05:
This really should go into the APM Forum so I'll close it here.  http://ardupilot.com/forum/viewforum.php?f=111

I'll still have a look at the logs though.

Sorry but if we're not strict about this our enhancements list turns into a support form.

closed 2015-10-05 by rmackay9

rmackay9, 2015-10-05:
It looks like when the vehicle is rotated i gets a large compass variance.  I suspect you can see this when it's on the ground as well (the beta MP has a real-time EKF viewer).  I'm not sure of the cause so I'll ask Paul if he has any ideas.
It's most likely some kind of frame issue rather than a software issue so if you know of any metal objects near the compass that could be the cause.

A couple of unrelated things I noticed:
- COMPASS_LEARN is "1".  This doesn't actually do anything in Copter-3.3 so might as well leave it as "0".
- the radio failsafe is not enabled.  Unless this is one of the first flights of the vehicle that should be setup.