ArduPilot/ardupilot #2939: EKF2 not initialising on boot, never recovers

dave-dave, 2015-10-05:
"PreArm: NavEKF not allocated" message shown on boot when using EK2_ENABLE=1.
It never recovers - autopilot works pretty well with AHRS_EKF_TYPE=1, though  AHRS arming problems are normal for me.
Using current HEAD.
Hardware is UBlox GPS, analog airspeed, lidar, Pixhawk and a plane with enough glue (and tape) to cushion any impact.
@priseborough  is there other details that you want/need?

priseborough, 2015-10-05:
On a pixhawk you will need to turn off terrain to free up enough memory via TERRAIN_ENABLE = 0

closed 2015-10-05 by dave-dave

magicrub, 2015-10-05:
Paul,

It may be helpful in the short term that of terrain is ground endangered to
put in a GCS message that says to disable terrain when ekf2 is enabled.