ArduPilot/ardupilot #2956: Use MAV_SEVERITY_INFO on appropriate messages

DonLakeFlyer, 2015-10-08:
There are a number of messages which seem like they should be MAV_SEVERITY_INFO. For example: when you request the parameter list the firmware sends out mavlink messages with various firmware version info set to MAV_SEVERITY_WARNING. This in turn causes QGroundControl to thinks these are errors since they are marked as warning. Shouldn't these be MAV_SEVERITY_INFO? There may be others. I'm just getting started with APM Flight Stack support in QGC. This is what I've found so far.

DonLakeFlyer, 2015-10-08:
@rmackay9 This time I put it in the write palce. Can you assign to me?

rmackay9, 2015-10-08:
@DonLakeFlyer, wish I could!  No idea how to get new people to appear in the drop-down list.  Maybe once you do a PR or something it'll appear.

billbonney, 2015-10-12:
It seems the problem is this line https://github.com/diydrones/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L2025
Which changed from severity LOW to Severity Warning when that doesn't make sense for all that call it. For example when a waypoint is reached. 

BTW: You need to be a collaborator to get access to the labels etc !

rmackay9, 2015-10-12:
@billbonney, @DonLakeFlyer 
Ok, I think I remember Tridge trying to decide if we needed a MAV_SERVERITY parameter for that call or not.  At the time they were all sent with the same severity so we left it out.  I guess we need to add it.

ggregory8, 2015-10-20:
I propose changing to MAV_SEVERITY_INFO at least. As most use cases of this are for low priority messages. The problem we have now is all these 'info' messages, such as firmware version etc, are being displayed on Mission Planner HUD and translated to voice as they have a warning severity.

magicrub, 2015-10-20:
I agree with Glen

rmackay9, 2015-10-21:
I think adding a severity to the gcs_send_text_fmt message would be best.  That way we can specify what severity should be used for each message we want to send.

ggregory8, 2015-10-21:
Ok that sounds like the best plan. Will it make 3.3.1? Otherwise this annoying issue could be around for awhile until next release.

rmackay9, 2015-10-21:
No real changes besides the one bug fix are in 3.3.1 I'm afraid so this won't make it.

closed 2015-11-16 by squilter