ArduPilot/ardupilot #2965: ROI ignored when switching from Loiter to Auto

billbonney, 2015-10-09:
[2015-10-09, 14:58:35] Bill Bonney: I just had a SITL mission with two ROI set, stop looking at the ROIs
[2015-10-09, 14:59:19] Bill Bonney: when I reuploaded the mission, no changes, it went back top looking at the ROIs
[2015-10-09, 15:00:13] Bill Bonney: And I can reproduce
[2015-10-09, 15:01:04] Bill Bonney: Just switch to Loiter (while in auto) then back to Auto.

I think this issue is related to the fact that when you 'pause' a mission by going to loiter and the ROI command has passed, it's not registered again until a new ROI command is encountered

I would have expected the mission behaviour to resume where I left off.

rmackay9, 2015-10-11:
This is a bit tough because it means storing more vehicle state.  The do-set-roi command could have been run long before the currently active nav command so we'd have to store it's state as we exit Auto.

magicrub, 2015-10-11:
So?

rmackay9, 2015-10-11:
So there!

magicrub, 2015-10-11:
>>... so we'd have to store it's state as we exit Auto
yeah, well, we should be!

magicrub, 2015-10-11:
What I mean is, it would be nice to unify the state of the whole mission
system in a way that when we exit auto, we can backup (via memcpy?) the
struct/class and then copy it back on a resume. We're doing close to that
all over the place but doing it in one place would be cleaner and less
prone to bugs such as his ROI one.

billbonney, 2015-10-11:
I think it's important to being able to 'pause' and 'resume' a mission as expected by the consumer. It's not what i expected in behaviour. 

As you said, the ROI could have been run long before the current waypoint, and is in lots of cases, as it's probably the 1st or 2nd on the list as only one ROI would have been set. You'd have to restart the mission to get the result required.

You only need to store it while active (which it must do anyway), doesn't power off and on restart the mission in anywise (hence that other request to not do that in battery swap?)

R-Lefebvre, 2015-10-12:
I think it's important to have this functionality.  Not just for Loiter, but for the Stop mode.  This feature is something that some other advanced commercial flight controllers have. User plots a mission, something goes not according to plan (could even be birds in the area, need to take a quick break from monitoring progress of the mission, who knows) so they Stop the mission.  Then resume.

magicrub, 2019-04-30:
@Naterater no