ArduPilot/ardupilot #2974: New defaults for Tradheli 3.3

ultrafuge, 2015-10-11:
Hi,

helicopters need different tuning for roll and pitch axis, therefore I would like to suggest to change the default value of : 
RATE_PIT_P=0.060 (now 0.020)
RATE_PIT_D=0.003 (now 0.001)

I think it is easier to control 'tail wagging' caused by a too high RATE_YAW_I, than 'weathervaning', therefore RATE_YAW_I=0.5 (now 0.1) should be better.
Most important ACCEL_Z_P=0.5 never worked for me. Usually I use ACCEL_Z_P=0.3 to 0.4 therefore a default of ACCEL_Z_P=0.2 should be o.k.
Could you please add RATE_YAW_I and H_COLYAW to the TUNE options for channel 6. 
It would be nice to have reasonable heli limits in MP in the future as well.

Thank You

R-Lefebvre, 2015-10-12:
Thanks for entering this issue Ultrafuge, this is the best way to manage this work.

ChristopherOlson, 2017-10-27:
This issue was partially resolved by PR  #7127. Since AC3.3 parameters have changed, we have a new attitude controller, scaling is different, and we have a new tuning wiki specific to AC3.4 and later.

The defaults for helicopters can never be right for all helicopter models, as each one needs to be tuned individually. Do we still need to look at different defaults for TradHeli in light of all the changes since AC3.3? I do not believe we leave these issues open indefinitely.

OXINARF, 2017-11-01:
Copter 3.3 is now two releases old so any change proposed here is likely to not be correct anymore. Also, this issue is two years old, if the vehicle maintainers haven't changed the default values it's likely they don't want to do it.

For these reasons, I'm closing this.

@ultrafuge Feel free to open a new issue if you still think that current default values need some change to be better for all helis.

closed 2017-11-01 by OXINARF