ArduPilot/ardupilot #2975: MKBLCTRL ESCs dont work any more.

bellibelson, 2015-10-11:
I am happily flying with MKBLCTRL ESCs on version 3.2. With version 3.3 it doesnt work any more.

R-Lefebvre, 2015-10-12:
Hi,

Can you please explain was a MKBLCTRL ESC is?  I'm not familiar with that one.  And what exactly "doesn't work" mean?  Does it arm itself?  Doesn't run at all? Doesn't run well?

rmackay9, 2015-10-12:
So I know a little about this.  These are the microkopter ESCs and although it wasn't a supported configuration, there were some changes that could be made to the SD card to enable them.  As far as I know, nobody on the dev team has these ESCs.

It's probably something small that's happened to stop them from working though.

bellibelson, 2015-10-12:
Hi,

sorry, I will explain. mkblctrl are the ESCs from mikrokopter.de. A lot of people have them, Also you can exchange the FC on a copter from MK with a Pixhawk and then fly it with no other change. The ESCs are connected to the I2C bus, and a file has to be put on the SD card, then everything works very well with copter 3.2. More information is here: 
http://diydrones.com/forum/topics/upgrade-a-mk-into-a-arducopter

Regards

Derek

makuser, 2016-07-29:
@bellibelson have you tried the latest 3.3.3?

bellibelson, 2016-07-30:
Thanks,
but I have changed the ESCs now. I believe a lot of people here in Germany have these expensive ESCs laying around just like me, and could use them with Pixhawk.

makuser, 2016-07-30:
Okay, I understand... I have them too, that's why I ask.
Then I'll probably get the entire toolchain for PX4 and debug it, to find out where the problem lies.

peterbarker, 2021-07-30:
Well, I don't think we support these and haven't for many years.   These will be old and outdated by now.

I'm going to close this one - but if anybody things we really should support these please re-open.

closed 2021-07-30 by peterbarker