ArduPilot/ardupilot #2984: AP 3.4: minimum altitude fence still does not work.

marcmerlin, 2015-10-12:


marcmerlin, 2015-10-12:
documentation says FENCE_MINALT is in meters. If it actually is in centimeters, that would explain why it's not working :)

marcmerlin, 2015-10-12:
I'd provide you with a dataflash log posted on droneshare, but droneshare still fails to process 3.4 logs, and they get silently dropped.

gmorph, 2015-10-12:
FENCE_MINALT in ardupilot is in meters.  I know as I use it every weekend.

You FENCE_TOTAL is 0 indicating you have no fence points set hence no fence enabled.  You need to draw a fence in your GCS before you can enable it.

Thanks, Grant.

marcmerlin, 2015-10-12:
@gmorph  Ok, thanks for the info. I think we still have at least a documentation bug here.

I've used FENCE_ALT_MAX on copter which works anywhere by just setting it (it will cap your altitude without having to draw a fence).
I naively assumed that FENCE_MINALT in plane was its opposite, but it's not (and the 3DR support folks I asked didn't seem to know this either).

FENCE_MINALT in Plane does say  "Minimum altitude allowed before geofence triggers"
FENCE_ALT_MAX in Copter says "Maximum altitude allowed before geofence triggers"

The copter one works without you drawing a geofence box. You say the plane one does _not_ work without drawing a geofence box, and in both cases the documentation says they work the same, which from what you say, they don't.

My suggestion is to at least fix the plane doc. More generally I think we should unify how plane and copter behave here.

And would you say that in my case I need to draw a geofence box that is the size of the country I fly in to get around that box requirement since I don't care about the lateral box, but only about the altitude?

Thanks

gmorph, 2015-10-12:
Hi.  You are correct.  The FENCE operation in Copter is completely different to that in Plane.  I was referring to its operation in Plane.
This is the documentation for the plane geofencing
http://plane.ardupilot.com/wiki/geofencing/
I have added a note to the top clarifying that for Plane the polygon needs to be in place.
I am going to reopen this issue and change the title to reflect the request that copter and plane geofence should work the same way.
Thanks, Grant.

gmorph, 2015-10-12:
Related to
https://github.com/diydrones/ardupilot/issues/1383

marcmerlin, 2015-10-14:
Thanks much for updating the doc, very helpful.

marcmerlin, 2015-10-14:
Mmmh, I just realized that my trick of just making a geofence the size of the US so that I don't have to reprogram the fence of every location I go to if I only care about minimal alittude, is not going to work because I need a rally point it seems.
Is there a way to just have it use the RTL point if I skip providing the 'set return location' ?  The wiki doesn't seem to say either way.

gmorph, 2015-10-14:
I tested in SITL in MAVProxy and if you don't specify a rally point the plane will return to the HOME location which may vary slightly depending on how you did your takeoff.  However, MP won't let you upload your fence unless you specify the return location so this won't work.

In Plane we don't really want people having it return to their HOME position as this means it usually circles over the flight box and pits of a flying field which is normally very bad.

Thanks, Grant.

marcmerlin, 2015-10-15:
thanks for looking and your reply. For what it's worth, flying back to HOME is what RTL does, but I make sure to arm my plane somewhere reasonable and use a 100m RTL altitude.
Can't say whether allowing the same for geofencing should be allowed, or not, I'll leave that up to you.
In the meantime, it sounds that to get minimum altitude fence functionality in plane like I would get it in copter, one should
1) make a huge fence area
2) set the required return point
3) upload
4) go in the settings dump and remove the return point
(I'm documenting this here in case someone finds this bug later and is trying to do the same as me)

gmorph, 2015-10-15:
I haven't tried that but it will probably do what you want.
FYI I fly at a few different locations.  I have waypoints and fence points setup for each and I just load them when I get to that location.  Only takes a couple of minutes.  I guess if you fly at loads of different locations all over the place this could be a problem.

marcmerlin, 2015-10-15:
On Wed, Oct 14, 2015 at 05:11:18PM -0700, Grant Morphett wrote:
> I haven't tried that but it will probably do what you want.
> FYI I fly at a few different locations.  I have waypoints and fence points setup for each and I just load them when I get to that [...]

In my case it's more that I can't use my 3DR radio since it conflicts
with my LRS system, so I can't easily setup wirelessly, and it's a bit
of a pain to have to connect a laptop via a USB cable before each
flight.
Thanks again for the replies.
-- 
"A mouse is a device used to point at the xterm you want to type in" - A.S.R.
Microsoft is to operating systems ....
                                      .... what McDonalds is to gourmet cooking
Home page: http://marc.merlins.org/                         | PGP 1024R/763BE901

WickedShell, 2015-10-15:
You could set the rally point to be unreasonably far away so that it never
passes the distance limit check. Just never get to close to it, and ignore
the warning on initial connect. But for that reason your method of
uploading and deleting send reasonable.

marcmerlin, 2015-10-31:
@gmorph and @WickedShell I wanted to report back on my testing.
Sadly, I was unable to setup any geofencing with apmplanner (the latest rc). Right clicking on the map only gives me "go to here (guided)" "go to here (alt guided)" and "point camera here"
Is this a known bug, or documentation shortfall, or am I just unable to use mission planner?

I tried tower on android next, and it's worse. Tower comes with no documentation on what the icons mean, or whether you can setup geofencing. At least no documentation that google could find. Any idea who to file this with if tower is supposed to support geofencing?
https://github.com/DroidPlanner/Tower/wiki is not helpful
neither is https://3drobotics.com/kb/tower-best-practices/ :(

I ended up having to use virtualbox and mission planner. That one made it easy to do geofencing and it actually matched the documentation on http://plane.ardupilot.com/wiki/geofencing/
And yet, altitude geofencing wasn't working.
I found out in the end that min alt geofencing will not work if max alt geofencing is not set (i.e. 0). That seems lke a bug to me since the documentation implies you can have max alt set to 0
Namely the plane docs say "FENCE_MAXALT < FENCE_MINALT AND FENCE_RETALT is <= 0 then ALT_HOLD_RTL is the altitude used on a fence breach."

In the end , I got it to work once I set FENCE_MAXALT, but I didn't want to have it set.
Also, I didn't find how to remove the fence return point (since I don't want to use that, but want RTL home to be used instead). Do you know how to do that?
Thanks

marcmerlin, 2015-11-03:
I did more testing 2 days ago, and here's what I did:
1) setup a geofence the size of the US
2) setup the geofence return point in the middle of the atlantic
3) FENCE_RET_RALLY 1 

This allows ardupilot to use the RTL point as a fence breach return point and it will go there if I fly too low.
However, I was still forced to set FENCE_MAXALT for FENCE_MINALT to work, and this is neither documented, nor really desirable IMO.
Can you put me in touch with someone who can edit documentation so that we can at least document the current behaviour properly ?
(and also either fix the docs on how to geofence with apmplanner, or state that mission planner is requred for now)

rmackay9, 2015-11-04:
@marcmerlin, I haven't read all the discussion above but in case it helps, the Rally library has been enhanced recently to have an RALLY_INCL_HOME parameter which, if set to "1" will mean that home is always included as a location to return to (if it's closes).

marcmerlin, 2015-11-04:
Thanks @rmackay9 . In my case not needed since I have 0 rally points and I was only trying to set a minimum altitude deck, but this sure sent me in a rabbit hole :)
Thankfully the code properly ignores my fence return point 4000 miles away and uses my RTL/HOME point as a destination for GUIDED when altitude fence kicks in.
However this is really so much more complicated with AP. With AC, you only set the altitude limit and it RTLs on breach. None of the complicated stuff I had to do to get this with AP. Hopefully AP will make this easier in the future but in the meantime I'd like to document the state of affairs in case others would like to benefit from the existing functionality.

marcmerlin, 2015-11-07:
Short of better documentation, I wrote my own for now:
http://marc.merlins.org/perso/rc/post_2015-11-06_Using-Ardupilot-Geofencing-As-An-Altitude-Deck.html

marcmerlin, 2016-02-02:
Hi @tridge in addition to allowing a mix and man altitude for geofence without requiring a huge fence, which seems to trigger float errors in your simulator

marcmerlin, 2016-07-06:
If anyone finds this issue later, this is now the setup I have:
FENCE_MINALT 50
FENCE_MAXALT 300 (required or min won't work)
FENCE_RET_RALLY 1 => go back to nearest rally point or home instead of fence point which I put way far in the ocean
FENCE_ACTION 1
FENCE_RETALT 100
RALLY_INCL_HOME 0 => go to rally point instead of home (i.e. don't include home in rally points), unless, see below:
RALLY_LIMIT_KM 0.2 => force rally to home if more than this distance from rally and home is closer

It's still kind of hackish, but works well enough for me. Hopefully this is useful to others.

amilcarlucas, 2021-03-09:
This has been implemented in #15288 and merged in master. Closing this one now

closed 2021-03-09 by amilcarlucas