ArduPilot/ardupilot #2986: ArduPilot does not do limits checking on most of it's parameters...

davidbuzz, 2015-10-13:
Currently the Min and Max values for parameters are only described in comment form, and are used by some GCS's for presentation of the data as a "slider" or similar, but they are NOT sanity checked in the flight-controller itself, making it high-risk that user/s can enter arbitrary incorrect information into their system.  
Typically with "value less" data like P I and D values that have weird range/s ( such as 0.01 -> 0.001 ) etc are easy to misplace the decimal point, and this can be catastrophic to flight performance, and it's just a typo.

davidbuzz, 2015-10-13:
This issue proposes then to determine an absolute "never exceed" maximum value for any flight platform, and an equivalent minimum value, and keep this data in the micro-controller with the flight code, and deny changing any parameter change-request where the parameter would exceed these acceptable limits.

WickedShell, 2016-02-09:
I just want to point out that there are parameters that are best worked with outside the ranges listed in the comments for somethings, especially if you have a vehicle that is bit of a hardware edgecase. (I'd have to go look at my tuning params to find which one, but I know I've had one of the values far outside the recommend range for a long time, but that really is the correct thing to do with it), so forcing the param to be within the currently listed ranges could be detrimental.

magicrub, 2016-02-09:
We have one that we regularly use outside of the range too.

Michael, What we really need to do is change the range on those params.

davidbuzz, 2021-08-16:
Just clarifying, This issue is a proposal to ENFORCE  a new never-go-below-MIN and new never-go-above-MAX for each param.  These will be potentially much smaller/larger than the existing range values ( which are just a suggestion and people often go outside of on unusual frames)