ArduPilot/ardupilot #2998: Antenna Tracker for APM/Pixhawk

artemen, 2015-10-14:
Continuous Pan Servos are expensive, however almost ANY regular servo can be EASILY, WITHOUT soldering gun/ programmer modified for continuous rotation use. Please, add such support at least for APM/Pixhawk based tracker. 
https://www.youtube.com/watch?v=XUm2nj7te8E

earsneyes, 2015-10-19:
Continuous Rotation Servos are already supported in the last version but Manual mode doesn't work with CR servos it think. 

Look for the SERVO_TYPE parameter.

0=Position - traditional standard servo
1=OnOff 
2=ContinuousRotation

artemen, 2015-10-20:
Well, in the sense of ability to cotrol those servos -yes, but yaw is still limited to 360 degrees by the YAW_RANGE max value 360.

Naterater, 2019-01-07:
Continuous servos have been available for antenna tracker for a few years now.

closed 2019-01-07 by Naterater