ArduPilot/ardupilot #2999: Refactoring I/O MCU API of RasPilot

raspilot, 2015-10-14:
Hi, @tridge @lucasdemarchi . As you known, RasPilot has a STM32 I/O MCU (derived from px4io) communicating with Raspberry Pi via SPI. The current API is crass, and does not support some features like safety switch, px4_mixer, etc. 

So I'm trying to refactory the API. And did some preliminary works. Here are some commits:
https://github.com/raspilot/ardupilot/commit/894f535eb2edcafa18dcd0fec9a88514116297b6
https://github.com/raspilot/ardupilot/commit/462d44f213e52994e7e1d4fc5c267b7254afe97b
https://github.com/raspilot/ardupilot/commit/7443c11fd2e99863bd34454647cf23bb8342212e

But that would take in too much modifications outside the Linux HAL. I wonder it would be a good way. And here are some ways come out of my mind:
1. Add a interface to AP_HAL, like the commits above do.
2. Add the API to LinuxUtil: iomcu_read(), iomcu_write(), ...
3. Add the API to HAL_Linux_Class
4. Add the driver into kernel, and leave a posix interface in Linux.

It would also be better for the API to be compatible with different hardware ports: SPI, I2C, or UART. Now it's via SPI because spi is higher in speed. But there would be a chance to change to other interface. Chances someone else would be interested to reuse the I/O MCU structure.

Any suggestion? Thanks!

IamPete1, 2021-08-06:
I'm going to close this since there has been no response, and the raspilot project as not had any recent changes.

closed 2021-08-06 by IamPete1

lucasdemarchi, 2021-08-09:
as raspilot is not supported anymore, anything mentioning it could be closed