ArduPilot/ardupilot #3000: FS EKF and  EKF GATE issue

roque-canales, 2015-10-14:
I set up a platform to test the EKF filter in the GPS management when it does not produce viable positions.

First test:
Test reactions in flight of drone LOITE mode when the bias of VEL and POS or exceeds 1. To do this I disabled FS EKF by setting it to 1 then I set VEL_GATE and POS_GATE reduced values (respectively 5 and 10, then 3 and 5, and 1 and 1).

According to the EKF manual in "copter.ardupilot.com", when bias values exceed 1, the EKF filter no longer takes into account theses values ....

In reality, nothing happens, it is clear EKF STATUS is in the red (so EKF detect issue), but the drone  goes left and then right louder until ANGLE MAX blocked in one direction.

In this case, the only way to recover the drone is Stabilize MODE ......

Second test:
Test FS_EKF toggle when the drone flight in LOITER mode, to do that I set FS_EKF_TRESH to 0.8 and then 0.6, Also I set FS_EKF_ACTION to 2 to switch to ALTHOLD mode in case of threshold.

Logically soon as one of the 5 BIAS EKF STATUS measures (VEL, HPOS, VPOS, COMPASS, GROUND) exceeds the value of FS EKF, FS EKF should switches into preset mode (in my case ALTHOLD ) to prevent:
The drone MAX ANGLE in one direction
Warn the pilot about the problem
give the hand to the pilot in a mode not using the GPS before it's too late (max angle in one direction)

Eventually, to avoid accidental triggering, FS EKF TRESH could be:
A Bias value to not exceeded (as is supposed to be)
Exceeding authorization period in milliseconds before toggle FS action.

We need to prevent that in the event of driver sends the drone in an area where the GPS is subject to reverberation (usually if it flies down the trees, or too ready of a building), avoid at all costs that the drone take max angle, because the pilot not have time to react, due to the fact that drone take big speed too quickly.....and crahk one the wall.

rmackay9, 2015-10-15:
@roque-canales, so I guess these are test you've done in which you've forced the innovations high?  It would be good if you could include a dataflash log.

roque-canales, 2015-10-15:
download logs here:
http://www.innovadrone.com/log.zip

roque-canales, 2015-10-19:
randy, have you checked shared logs following link that I gived to you?

rmackay9, 2015-10-19:
@roque-canales, I haven't yet.
By the way, could you answer my question above?

roque-canales, 2015-10-19:
Absolutely, I test flight with default ekf setting using gps position alterator (aluminium sheet attached to servo, in flight rotated over gps antenna) inducing high innovations ..

just after seen that with default setting innovations goes high but fs ekf no auto invoke fs ekf action, I reduced the pos and vel  gates values to magnify innovations.... Despite innovation enormous values, fs ekf never tresholded fs ekf action.....

So questions are:
There's multiplicator between innovations and efk theshold for vel and pos innovation?
There's actualy fs ekf activated only for mag innovations?
or more globaly, could you tell us how to? I can test all you want for enhance this part

roque-canales, 2015-10-25:
Randy, have you check the logs?

roque-canales, 2015-11-08:
Dev team,  we are able to test activated fs ekf for gps, waiting your new release for test it

magicrub, 2015-11-08:
So it's in master now and you're waiting for master to get released?

roque-canales, 2015-11-09:
for test it I need compiled beta version, so are you meaning that it's already available on released ac 3.3.1?
if it's not the case could you share it as new beta version through mission planner?

lostcontrol, 2015-11-09:
If it is in master, you can get a daily build from here: http://firmware.diydrones.com/Copter/latest/

roque-canales, 2015-11-17:
I just test it on 3.4dev, It seems that drone handle better low viable positions from gps.

Does glitch algo was disable on 3.3?

Today to trigger fs ekf from GPS issue, we need to wait 5 long seconds.

There's possible to add parameter in full parameter list to tune this delay ?

In five seconds drone can travel lot of distance before trigger to althold or land.
If user it's near from wall, it could be better that drone stop use gps and trigger failsafe for use regain control.

roque-canales, 2015-11-17:
I just make another few test this afternoon, and I succeded to trigger on ekf failsafe due to low viable position.

But FS algorithme seem's to be very complex. 

To trigger FS EKF due to low viable gps pos,  It is as if it must to have EKV VEL  + EKF POS + Gps GLITCH for more than 5s.

This cause that FS trigger only in case of GPS Disconnected, or very very bad GPS, and before gps is triggered, drone goes full lean angle in all directions, witch is dangerous in case of use near wall/trees.

Could you please change algorithm to able each one (EKF VEL /  EKF POS / Gps glitch) to trigger independently FS EKF? also it could be interesting to set for each one different delay, for tune each trigger.

Naterater, 2019-01-07:
@roque-canales, is this still an issue with EKF3?

rmackay9, 2019-01-07:
EKF2 is really the default so if this is still a problem with EKF2 then we should keep it open.

IamPete1, 2021-09-12:
Any updates in this issue? EKF2 is no longer the default....

rmackay9, 2021-09-13:
I think we can close this.  I'm not aware of situations on real vehicles where the EKF failsafe is not catching problems and we've improved the check since this issue was raised in 2015.

closed 2021-09-13 by rmackay9