ArduPilot/ardupilot #3001: bulding master for ArduPlane and ArduCopter fails on OSX with SITL , but with different errors

davidbuzz, 2015-10-14:
here's the output of: "make sitl 2>&1 | grep error" for each of them:

Plane SITL build fails with: 

Log.cpp:330:24: fatal error: non-constant-expression cannot be narrowed from type 'enum safety_state' to 'uint8_t' (aka 'unsigned char') in initializer list [-Wc++11-narrowing]

Copter SITL build fails with: 

ardupilot/libraries/SITL/SIM_Tracker.cpp:97:9: error: variable 'pitch_rate' is uninitialized when used here [-Werror,-Wuninitialized]
ardupilot/libraries/SITL/SIM_Tracker.cpp:91:9: error: variable 'yaw_rate' is uninitialized when used here [-Werror,-Wuninitialized]

joseluisUEX, 2016-01-13:
I'm having the same problem, did you find a solution?

davidbuzz, 2016-01-13:
nope.  you might want to see if a different version of gcc helps  ( ie use gcc from 'brew', not osx ) ..?

joseluisUEX, 2016-01-14:
Finally, I solved the problem and I can make the project using the terminal. However, when I try to do the same in Eclipse, I get the following error:
make: arm-none-eabi-size: No such file or directory
make: *** [px4-v2] Error 1

Do you know how to solve it?

Regards
Jose Luis

OXINARF, 2017-06-02:
I believe this is no longer true and SITL can be built on OSX.

@davidbuzz @joseluisUEX?

davidbuzz, 2017-06-02:
lets close this, and assume it's fixed.    I can re-open in the future if that ends up not being the case.

closed 2017-06-02 by davidbuzz