ArduPilot/ardupilot #3014: Don't allow mode changes if in a RC failsafe if the mode requires RC input

WickedShell, 2015-10-16:
We currently allow you to change to modes that require RC input even when we know you don't have RC input due to a RC failsafe being active. Since there is no valid RC input changing to any of these modes could be surprising as the user would have expected to have control.

For ground testing and sanity, we would only run this check when armed, when disarmed you could swap to any manual mode quite safely.

For plane the modes I can think of that would be affected:
  -Manual
  -Stabilize
  -FBWA
  -FBWB
  -Autotune
  -Acro
  -Cruise

For consistency we should try to make all the vehicles behave the same.

tridge, 2015-10-16:
as we discussed on mumble, I think this is OK as long as we use the rc_failsafe_active() function, which will return false if throttle failsafe is disabled

WickedShell, 2017-06-02:
I've apparently reversed my position on this as has @tridge as we discussed this extensively on the dev call last week and was decided against it, and that the current behaviour has a lot of utility. Closing as the conclusion was that we actually want to allow this behavior as it is quite useful in emergency situations, or in preflights. GCS's should be monitoring/and prompting the user in this situation instead.

closed 2017-06-02 by WickedShell