ArduPilot/ardupilot #3017: Copter: Tracking Why HIL Is Broken

majorpr13, 2015-10-16:
This is an issue and a question that may not be obvious. The theory behind why HIL is broken currently I believe is because of the addition of inertial navigation. My question is then what exactly from the Inertial Navigation broke HIL? I would like to help in fixing this as there is a lot of demand to make HIL possible for both Plane and Copter code bases. Is anyone on the development branches working on this and would like help?

rmackay9, 2015-10-18:
I think that none of the simulators could provide correct accelerometer data.  I only personally tested with AeroSim so perhaps others will work better.
http://www.aerosimrc.com/en/home.htm

majorpr13, 2015-10-19:
Randy, 
I think I can work around that as I have a custom model I am using. Could you please describe however to get the arducopter to accept the HIL MAVLINK commands? It appears you might be able to just do a build with the flag in the arducopter config file...Is this all that is needed? Thanks.

patnolan33, 2015-10-19:
I have been working on this for a little while now, and I seem to be spinning my wheels here. I posted a topic or two on DIY Drones about my progress: http://diydrones.com/forum/topics/custom-hil-program-with-ros-gazebo-and-ardupilot?xg_source=activity

I can't seem to get the arducopter code to respond to any HIL MAVLINK commands, similar to @majorpr13. I would love to get some semblance of HIL working as it would provide for a very useful test environment.

lennoncork, 2015-11-06:
@patnolan33 I've just started to look at this myself. I had a quick look at your post, and at the MAVLINK messages here. https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml. here state that the HIL_STATE message is depreciated. Not sure if it's your issue, but worth a look. I'm also assuming you've uploaded the HIL firmware to the Pixhawk (i.e. from mission planner).

patnolan33, 2015-11-06:
@lennoncork I actually went away from ArduCopter for the moment in favor of the PX4 firmware. I was able to get HIL working (somewhat) with that flight stack using a combination of mavros (http://wiki.ros.org/mavros) and jMAVSim (https://pixhawk.org/dev/hil/jmavsim). For my immediate purposes, this seems to be working fine. That being said, I would prefer to use ArduCopter because I am more familiar with it.

patnolan33, 2015-11-06:
My apologies, I though I addressed all parts of your question @lennoncork. I did see that the HIL_STATE message is deprecated, so I did switch over to the HIL_STATE_QUATERNION. I still didn't have any luck with the ArduCopter platform. That being said, I wasn't even able to get the Pixhawk into a state where it would act on the HIL_STATE_QUATERNION message. In other words, using the SET_MODE command or sending a COMMAND_LONG message with the command MAV_CMD_DO_SET_MODE, I wasn't able to get the Pixhawk into HIL mode. With the PX4 firmware loaded, the SET_MODE command works as anticipated and I am able to send HIL messages and receive the correct response. 

Assuming I could get the Pixhawk into a HIL mode, the HIL_STATE_QUATERNION should work. So I went ahead and built the HIL binary from the source code and still no luck. Feel free to ask any questions, I'd be happy to clarify anything that is unclear.

patnolan33, 2015-11-06:
One more clarification: When I loaded the HIL firmware directly, I verified that HIL state was in fact loaded by changing the attitude of the Pixhawk and seeing that the attitude in Mission Planner remained at 0 (as it should have). So even in a HIL state, the Pixhawk would not respond to any HIL messages.

OXINARF, 2017-06-06:
HIL is mostly broken and not considered fundamental anymore since SITL is a much better simulation environment. In any case #508 is open as an enhancement for someone that wishes to fix it.

closed 2017-06-06 by OXINARF