ArduPilot/ardupilot #3023: Copter should have RTL_SPEED parameter to decouple RTL and navigation speed

jschall, 2015-10-17:
Solo implementation: https://github.com/3drobotics/ardupilot-solo/compare/b43b2e532fb634811c43f1ad66cc494c90060070%5E...b43b2e532fb634811c43f1ad66cc494c90060070

marcmerlin, 2015-11-03:
I agree, I had the same problem when flying X8, its default RTL speed was stupidly slow and down right unsafe if you ever set low battery action to RTL instead of land.
I also had to up the vertical speed bigtime so that it would come down a lot more quickly than it does by default (until you're close to the ground, and then you can slow down).

jschall, 2015-11-03:
Exactly. I'd love if RTL were completely independently configurable down to acceleration limits. For Solo, we wanted RTL to fly home fast but accelerate quite softly while preserving a different behavior for GUIDED and AUTO modes.

rmackay9, 2015-11-04:
I think the two arguments are separate.
A test of this is, @marcmerlin would you also like the auto missions faster by default (maybe you don't use Auto much?).

marcmerlin, 2015-11-04:
@rmackay9 I indeed don't use auto, but when RTL kicks in as failsafe, it must come back as fast as possible and safe. In my case, I did indeed increase all the navigation speeds to the max since it was the only way to get that.
For what it's worth, I did have an X8 that was too slow to fly back and was destroyed as its battery died at 2/3rd capacity used while the X8 was RTL'ing. The battery was at fault (it was fully charged and yet died at 68% used), but if RTL had been faster, the X8 would have been on the ground instead of 100m in the air and descending over the landing point, when the battery died.

rmackay9, 2015-11-04:
@marcmerlin, ok, so we can certainly increase the default WPNAV_SPEED from 500 (just 5m/s) to perhaps 1000 (10m/s).
There's a separate to-do item about changing the battery failsafe so that it calculates how much power it needs to go home and ensure that it doesn't go over that limit.  This likely wouldn't have saved the X8 because it's likely it wouldn't have know that the battery was failing.

jschall, 2015-11-04:
@rmackay9 Still, I think that fundamentally it is better to consider RTL distinct from WPNAV. For a refined UX on a vehicle like Solo, it is definitely needed.

closed 2015-11-30 by jschall