ArduPilot/ardupilot #3026: If RC failsafe is active, EKF failsafe should not switch to ALT_HOLD

jschall, 2015-10-17:
There's perhaps a more general issue here, in that it is possible to enter a manual mode at all when RC failsafe is active. Solo occasionally runs into race conditions where a MAVLink mode change gets through after an RC failsafe is triggered and it can potentially cause Solo to drift away.

Solo implementation:
https://github.com/3drobotics/ardupilot-solo/compare/08ae8333a6c72c7076dc846fe292b402db2ff121%5E...08ae8333a6c72c7076dc846fe292b402db2ff121

rmackay9, 2015-10-18:
@WickedShell brought this up as well and raised an issue 2 days ago so closing this one.
https://github.com/diydrones/ardupilot/issues/3014

closed 2015-10-18 by rmackay9

rmackay9, 2015-10-18:
@jschall, I've pulled the Solo fix into master.
https://github.com/diydrones/ardupilot/commit/598925f8d2dfb1d42a43e4e22a6c335eaa3f81e7